TheCrypt0/yi-hack-v4 #2: MQTT Integration

fbrinker, 2019-01-18:
Nice to see the yi-hack-4 coming alive :)

Just read about the Telegram integration and looked into it.
The basic functionality of that script is to look for alarms and then to trigger something, in that case a Telegram message. I built the same for MQTT -> https://github.com/fbrinker/yi-hack-mqtt

So, as an idea:
Implementing an "alarm-watcher" calling hooks like publishing Telegram or MQTT messages or maybe both. And with MQTT integrated, the following (home) automation possibilities are almost endless :)

I guess the only problem is the limited space on the cameras, so people may have to place the MQTT binaries on the sdcard?

If the Telegram PR is merged into this project, I could look into integrating MQTT additionally, if people are interested in this. Otherwise I will update my project to the yi-hack-v4, if necessary :)

lance36, 2019-01-19:
Very nice, awesome feature to have for integration with home-assistant.

pltanton, 2019-01-21:
It also can be handy to control camera over MQTT. As far as I know it is not a big deal and can be done by touching of some files on camera fs?

TheCrypt0, 2019-01-21:
@fbrinker 
Hi, thanks for the suggestion! I'm not too familiar with MTTQ but I'll have a look on your project.

@pltanton 
What do you mean by "controlling" the camera? There are surely some things that can be done but most of the settings available from the app can't be changed directly on the camera at the moment.

fbrinker, 2019-01-22:
@TheCrypt0 just think of it as another (IoT) messaging client that other devices can subscribe to :)

@TheCrypt0 @pltanton I guess he wants to stop and start the motion detection remotely. I've tried to accomplish that in my project by killing and restarting some services, it almost works, but from time to time it seems to re-enable itself.
I don't want to trigger recordings while I'm at home, so my target was to enable it automatically when I'm leaving the house and to disable it when coming home :)

pltanton, 2019-01-22:
`I guess he wants to stop and start the motion detection remotely`
Yes. As well as toggle blue light and whatever what accessible directly from camera.

gaggi, 2019-01-22:
while working on the web interface I've looked around a little bit and stumbled upon this topic on the yi-hack-v3 firmware: [https://github.com/shadow-1/yi-hack-v3/issues/207](url)

After a little playing around I was able to control activity detection, the night vision IR light, status light and microphone by sending the following messages to the ipc_dispatch message queue:
`Status Light
on 01:00:00:00:08:00:00:00:76:00:01:00:00:00:00:00
off: 01:00:00:00:08:00:00:00:77:00:01:00:00:00:00:00

Nigthvision
on: 02:00:00:00:08:00:00:00:24:10:01:00:04:00:00:00:01:00:00:00
off: 02:00:00:00:08:00:00:00:24:10:01:00:04:00:00:00:02:00:00:00

Microphone
on: 02:00:00:00:08:00:00:00:2A:10:01:00:04:00:00:00:64:00:00:00
off: 02:00:00:00:08:00:00:00:2A:10:01:00:04:00:00:00:65:00:00:00

Activity 
on: 01:00:00:00:08:00:00:00:78:00:01:00:00:00:00:00
off: 01:00:00:00:08:00:00:00:79:00:01:00:00:00:00:00`

from reading the mentioned topic it should also be possible to do more things like control the motion of the dome cameras and so on.

TheCrypt0, 2019-05-11:
MQTT has been added in the last release ([v0.2.0](https://github.com/TheCrypt0/yi-hack-v4/releases/tag/v0.2.0)).

Closing

closed 2019-05-11 by TheCrypt0

janfajessen, 2022-02-27:
Great job!  I have the older firmware on the Yi Dome but would like to be able to control the movement or rotation of the camera with MQTT for Home Assistant.  It is not clear to me if it is only from the web ui or also with MQTT.  Is it possible now with this firmware?