TheCrypt0/yi-hack-v4 #4: Yi 1080p Cloud Dome

phoxy84, 2019-01-31:
I have this camera and are willing to help. My technical knowledge is limited but I am willing to help and learn :)

Also, camera is not working now (outside China problem) and I hope it will finally work

closed 2019-02-03 by TheCrypt0

Thommy3BG, 2019-03-01:
I own this camera too (european version).
If you need someone for testing... Please feel free...

hcrayfish, 2019-07-23:
> I have this camera and are willing to help. My technical knowledge is limited but I am willing to help and learn :)
> 
> Also, camera is not working now (outside China problem) and I hope it will finally work

hi man, i am in NL as well, but my xiaoyi 1080 p is just not working, can I ask how did you do so?

phoxy84, 2019-07-24:
I have to yi 1080p cloud cam with yi hack v4, that’s works for me in The Netherlands. Don’t know if it works on you camera to, you can join the discord channel mentioned above and ask there.

hcrayfish, 2019-07-24:
how did you configure the proxy do you know? i didnt understand how to do
that

phoxy84, 2019-07-24:
https://github.com/TheCrypt0/yi-hack-v4/issues/37

hcrayfish, 2019-07-24:
Thanks Alex for your swift reply. I have seen this post but I didnt know
where could I find  proxychains-ng file. Do you happen to know this?

phoxy84, 2019-07-24:
Please read the topic:
Same question: https://github.com/TheCrypt0/yi-hack-v4/issues/37#issuecomment-484697968

Answer:
https://github.com/TheCrypt0/yi-hack-v4/issues/37#issuecomment-484726170

So, ssh into your camera and follow the instructions from the topic.

It can be that you need a few reboots before it connect. Proxy's can be found online

hcrayfish, 2019-07-24:
> Same question: #37 (comment)

This is the exact party I did not understand.
Where do i put this? vi /home/yi-hack-v4/etc/proxychains.conf
Is it on my PC or on the webpage? I do see this
https://github.com/TheCrypt0/yi-hack-v4/issues/37#issue-431468853
but i do not find any where i could configure the proxy
can you share with me the screenshot where did you enter the command? (vi
/home/yi-hack-v4/etc/proxychains.conf)

phoxy84, 2019-07-24:
Did you ssh into your camera? Did you install the Yi hack v4?

hcrayfish, 2019-07-24:
Hi Alex,
I flashed my firmware successfully and see the screen
<https://user-images.githubusercontent.com/38787609/55877577-0ef14300-5ba3-11e9-9ea4-edcda6d5c1f2.png>
not sure what do you mean by ssh? what does it stand for?

phoxy84, 2019-07-24:
with ssh you can get into your camera, use a program like putty. Then you can change the proxychain file. There is a lot of info in the wiki and issues. Also connect with discord to get answers. and ask questions. In the discontinued yi hack v3 is also a lot of info (proxychain is different compared to yi hack v4)

hcrayfish, 2019-07-24:
i am using yi hack v4, and it just reporting "this camera can only be used
in China"

hcrayfish, 2019-07-24:
i am trying to use ssh to connect with the camera, however, it always ask
me to enter a password. my understand is the password is empty, so i just
hit the enter, but it does not work. do you know how to solve this?

phoxy84, 2019-07-25:
No, sorry. What are your settings?

hcrayfish, 2019-07-25:
iti s the default yi hack  v4, i have done nothing else. I do know how to
connect via ssh. It always ask me for the password. I asked in the discord,
and nobody replys. Could you may be help?

phoxy84, 2019-07-26:
What are your putty settings, what do you setup there?

hcrayfish, 2019-07-26:
i am using the mac/OS default terminal and thought the ssh is embedded by
default. dont know if i could configure that? or should i download the
putty for mac os?

phoxy84, 2019-07-26:
Via terminal you can get into it easy bij ssh to root@[ip of your camera]

so: ssh root@[ip of camera]

hcrayfish, 2019-07-26:
hi Alex, you are my hero. I have edited my configuration file even it is
still does not work. it is progressing.
My file configuration is as follows; not sure what is the protocol the
camera using, socks5, socks4 or http? do you know any good free proxy
website i coud look up?

random_chain

chain_len = 1

tcp_read_time_out 15000

tcp_connect_time_out 8000

[ProxyList]

socks5 150.109.195.10 1080

http 121.40.90.189 8001

phoxy84, 2019-07-27:
Just Google Chinese proxies, like http://free-proxy.cz/en/proxylist/country/CN/all/ping/all or http://spys.one/free-proxy-list/CN/ fill in a bunch, like 10 or 20 from different kinds, so socks4, socks5 and HTTP. Restart and hope for the best, maybe restart a couple of times if it does not work.

Also, make sure u have the latest yi hack, so V0.2.0 and have enabled the proxychain in the web status page of your camera. Also be sure you have the right yi app, so the Chinese one, read on the yi hack v3 (so not v4) github page for more info.