TheCrypt0/yi-hack-v4 #7: Home Assistant Integration Workaround

aptonline, 2019-02-05:
You may, or may not, know that for newer version of the v3 hack integration into Home Assistant wasn’t possible due to a bug in pureftp (https://www.home-assistant.io/components/camera.yi/). There has been a workaround posted (https://github.com/shadow-1/yi-hack-v3/issues/129#issuecomment-361723075) and I woundered if this could be taken into account with the v4 release?

BlueBlueBlob, 2019-02-05:
You will not need it with rtsp : better video quality and better performance.

aptonline, 2019-02-05:
Agreed, is RTSP likely to be included in any initial release of v4 or could the above work around be added as a stopgap until it’s included?

TheCrypt0, 2019-02-05:
Hi @aptonline,
the RTSP will be included in one of the first releases of the firmware.

The workaround for Home Assistant integration uses busybox's integrated ftpd instead of pure-ftpd, however I've had some problems with the former. As @BlueBlueBlob said, when RTSP will be included and supported in the public releases there won't be any need for the fix.

BlueBlueBlob, 2019-02-06:
For what I read. People are asking for rtsp support on home assistant.
All workarounds will give you a bad quality, just my point of view :/

schroederhaus, 2019-02-20:
Is rtsp in the current rc1 ? I just installed it for my Yi Home 720 and everything is working so far, just can not find the rtsp stream

aptonline, 2019-02-21:
The rtsp is only available to supporters of the project at the moment. Check discord for details. It works well but is still undergoing testing.

closed 2019-03-21 by TheCrypt0