TheCrypt0/yi-hack-v4 #9: Support for native Android app

lucaventurini, 2019-02-11:
Hello,

I have noticed that many on the Android Play Store lament the fact that the original Yi-Home app does not show notifications anymore, and I'm one of the unlucky ones.

Does this firmware somehow fix this situation? I know it is some magic that I'm asking about, but who knows, maybe it is just a matter of reverting something back...

Alternatively, what would it be a workaround that can be achieved today? I have read of a Telegram plugin, but it seems not implemented yet, am I correct?

Kudos to @TheCrypt0 for the great work!

BlueBlueBlob, 2019-02-11:
Hi
Yi app notifications have nothing to do with the camera. 
For workarounds : curl is implemented, so you can already play with ifttt. I don't know about telegram bot, but if it uses http post and get, it will be easy to add.
I can help you for ifttt or gdrive. For telegram bot, go to discord channel :)

Best regards

lucaventurini, 2019-02-13:
IFTTT would be great! Should I follow the instructions from yi-hack-v3 to flash the firmware?
If you can give me some advice to integrate ifttt I'd really appreciate that.

BlueBlueBlob, 2019-02-13:
I don't know, are you comfortable with bash ?
Here documentation for ifttt : https://maker.ifttt.com/use/bwOewRzXW_EYTrQv_zDyNf

With this you can send notification and last snap url.

Ask if you need help. I have to remake my ifttt script for motioneye.
And next release will include mqtt integration.

Have fun :)

lucaventurini, 2019-02-18:
Hi @BlueBlueBlob, do you have any script to share perhaps? I don't really know where to start, I'm ok with bash but don't want to reinvent the wheel ;)

And, as for the last snap url, is it a private url to the camera or a public one to the cloud?

TheCrypt0, 2019-03-21:
@lucaventurini

Unfortunately we cannot control from the camera how often the App will send out notifications. MQTT integration is being developed to allow to subscribe to motion events.

closed 2019-03-21 by TheCrypt0