TheCrypt0/yi-hack-v4 #11: How to install Hi3518E_SDK Helpful Hints

densanki, 2019-02-15:
Hi Developers,

if someone want to try to compile this project I recommand to use this beautiful helper article for installing toolchain and osdrv for Hi3518E development.
It includes detail error description and detail fixing.

http://www.programmersought.com/article/933051736/

I hope a will reduce of night work to get the setup done, I my case it was a very long night.

Addtional hint that was not on the article for Ubuntu 18.04
Errormessage
fatal error: 'uuid/uuid.h' file not found
Solution:
apt-get install uuid-dev

BlueBlueBlob, 2019-02-15:
Another trick if you use x64 OS : you need to add libs to run x32 apps on x64

TheCrypt0, 2019-03-21:
Thanks @densanki for the link! 

I think it was added to the Wiki by another user.

closed 2019-03-21 by TheCrypt0