TheCrypt0/yi-hack-v4 #12: 5F Models

major153, 2019-02-25:
First off thank you Crypt0 to reviving the project. I'm excited to see v4!

I have a 5F series of the Yi Home Camera - does this run the Hi3518e chipset?

It was mentioned that the 4F series uses a different chipset:
[https://github.com/shadow-1/yi-hack-v3/issues/233#issuecomment-436081622]( 4F chipset)

Anyway I can contribute to add support to a new camera?

leioo2000, 2019-03-03:
I think not, 5FCN runs the Mstar MSC313E chipset. I have also a 5FCN 1080p home camera 2 at home but so far no one can hack it. Actually all new Yi cameras that manufactured since 2018 June for Chinese market are all with the new chipset, I guess.

TheCrypt0, 2019-03-03:
Hi @major153
Sorry for the late reply. I've been busy for the last month working hard on the RTSP implementation. 

Anyway as @leioo2000 mentioned, they run on a different chipset. I would need to have look into the sdk and toolchain components as well as having one of the camera on hands.

I don't exclude that it could be possible in the future.

Greetings,
Crypto

closed 2019-03-03 by TheCrypt0