TheCrypt0/yi-hack-v4 #17: Yi Home 1080P bricked after firmware install

mvereecken, 2019-03-14:
I bought a Yi Home 1080P (48USYJH1YF) yesterday, configured it, all worked well. Then I tried to install the firmware from https://github.com/shadow-1/yi-hack-v3 1 (by following the instructions). For my camera this had to be rootfs_y20 and homey20. I got a orange blinking light, then a solid orange one. This is not good.
Then I tried a couple of scenarios:

- Put the official firmware back on the SD card
- Tried the recovery file from https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/23494084746

Finally, I found your repository and tried to use it. The orange light blinks for a while at first, so there's definitely something going on, then it becomes solid orange after a while (a minute or so).

SD card was MSDOSFAT32 formatted, there are no file extensions in the filenames, I reset the camera a couple of times,… So I’m out of ideas of what I can do now.
So, the camera is bricked for the moment. Any idea how I can unbrick it?

TheCrypt0, 2019-03-17:
@mvereecken 
I've your same model and I'm not having issues with flashing. 

Can you confirm that you can't revert back to stock using shadow's images?

ech0wav3, 2019-03-17:
This has also happened to me with 2 separate cameras now. Both are Yi Home 1080P's. The DIDs from each camera are slightly different and start with 48USYBCRZS and 48USY8K3EW. I purchased both via Amazon.ca. I didn't check the second one I purchased, but the first one had no newer firmware available out of the box. (According to the Yi website, the newest firmware was released in December 2018, IIRC.)

I tried putting custom firmware on both cameras via shadow-1's yi-hack-v3 (specifically the prebeta release without PureFTPd). Subsequently both cameras became bricked.

As mvereecken did, I also tried flashing back stock firmware from the site and using shadow-1's recovery images without success. I also tried newer and older versions of shadow-1's firmware to no avail. Finally I tried the custom firmware from TheCrypt0's (this) repository without any change.

Each time I tried a different firmware, the camera would seem to detect it, blink for various amounts of time and rapidness (depending on what firmware was being flashed) and then flip to a steady orange light and do nothing else. Rebooting the camera, taking out the microSD card and pushing the reset button all had no effect in getting the camera to boot properly.

If I had to guess, I'd say either the camera doesn't like its firmware being downgraded or the newer firmware somehow prevents custom firmware (either on purpose or accidentally via the camera's native file system changing or something).

If you have any suggestions, or I can be of any help providing more info to fix the bricking problem, let me know.

mvereecken, 2019-03-17:
Hi TheCrypt0,
> Can you confirm that you can't revert back to stock using shadow's images?
Yes, I can confirm. In fact, I tried it again, just to make sure. The light doesn't even blink any more, if I try to install the recovery version or the latest version of the manufacturer.
(Also, I bought a new micro-SD card, just to make sure that wasn't the problem here)

mvereecken, 2019-03-17:
ech0wav3, thanks for posting your description.

Fantoccini, 2019-03-21:
Out of curiosity, how did you find the DIDs ? `48USYJH1YF` and `48USYBCRZS`

ech0wav3, 2019-03-21:
@Fantoccini I just found mine from the label on the back of the box. However, it can also be found on the sticker on the back of the Home 1080p below the QR code. (The QR code, if scanned, has the same info plus a date code at the end. Presumably this is to indicate the manufacture date of the camera.)

TheCrypt0, 2019-03-24:
Hi,
I've been working with @loafdude on a fix for this situation and it should be fixed in the new 0.1.1 release.

https://github.com/TheCrypt0/yi-hack-v4/releases/tag/v0.1.1

Can you give it a try and let me know your results? 

Greetings

mvereecken, 2019-03-24:
@TheCrypt0 and @Loafdude: a million thank-you-very-muches!!!! You made my weekend! This indeed unbricked the camera successfully! I was able to install your firmware hack and the camera is fully functional again! Thank you thank you thankyou!! I really appreciate your help!

Two questions:
1) how can I donate to your project?
2) Can you elaborate on how to add this in Home Assistant? I added this:
```
camera:
  - platform: ffmpeg
    name: Wolfwatch
    input: -an -i rtsp://192.168.0.39/ch0_0.h264
```

ech0wav3, 2019-03-24:
Awesome work guys! I can confirm my camera is unbricked as well!

@Loafdude, thanks for doing the digging and figuring out the solution! @TheCrypt0, thank you for implementing the fix and getting it out so quickly.

I really appreciate the hard work both of you put in to solve this issue! :)

closed 2019-03-26 by mvereecken

seth79667, 2019-03-28:
That hack totally saved my Yi camera! Thanks!

v0ltage120, 2020-01-14:
Just got the same issue, I put hack-v4 firmware as mention, but I got blue-yellow-blue and nothing at booting... doesnt seems to flash... 
If I remove the SD card and boot I got solid yellow... :S

llellouche, 2020-06-01:
Thanks you save me ! After 3 hours and lots of tries your solution worked for me

axilic, 2020-11-16:
Hi. I bought a "Yi Dome Guard Camera" from aliexpress.com Yi Store. I had a connection problem. They sent me firmware to install. Camera was bricked while installing the firmware. Now only the red light is on and no sound is heard. Can you help me recover the camera? (IC: 20436-YRS3019)
@TheCrypt0 @ech0wav3

idanny76, 2021-05-31:
> 
> 
> Just got the same issue, I put hack-v4 firmware as mention, but I got blue-yellow-blue and nothing at booting... doesnt seems to flash...
> If I remove the SD card and boot I got solid yellow... :S

Did you solve this? I have Dome 720P with the same issue.