TheCrypt0/yi-hack-v4 #18: 22EU Yi home 720 support

xvolte, 2019-03-14:
Hello,

Does the firmware support the 22EU yi home camera ?

Thanks in advance

TheCrypt0, 2019-03-17:
Hi @xvolte,
Yes, they should be supported. 

Greetings

xvolte, 2019-03-18:
Looks like the firmware is not flashed.
the files are however present on the sd card in a fat32 image.

xvolte, 2019-03-18:
Maybe i should change the name of the files ?

xvolte, 2019-05-15:
Would it help if i could provide some info of the cameras ? i have a couple here at home!

xvolte, 2019-05-21:
@TheCrypt0 would you agree to look into it if i send you one for free?
I have unfortunately 4 cameras which are all 22EU. 
Yi-hack v3 works on them but i d like to have your features (rtsp, mqtt, homebridge integration, googlecloud features...)

TheCrypt0, 2019-05-21:
@xvolte 

Interesting that v3 works on them.. this project was a fork of shadow's version.
What home and rootfs files did you use to flash with v3?

kobytg17, 2019-09-02:
@xvolte ever got this to work? I have the same model

skarol, 2019-12-28:
@TheCrypt0 I also have a problem with Yi home 720 22EU. The installation process was smooth and correct (according to guide), but I'm not able to connect with WebUI. How can I check if the hack is installed or not?

hans-dot, 2020-03-05:
@skarol  any progress for YI home 22EU ? firmware is 1.8.6.1B2016040071154

skarol, 2020-03-06:
@hans-dot unfortunately not. I've installed back original firmware.

gburas, 2020-07-16:
@skarol How You installed back original firmware, because i have used yi-hack-v1 but video quality is so poor. I had 22EU version and after the hack camera started to talking in chinese and i can`t connect to it.

jonascoenen005, 2021-10-01:
@TheCrypt0 Any update on the 22EU? i'm also looking at issue https://github.com/TheCrypt0/yi-hack-v4/issues/26, but no update.. Still no solution for this?

baruchshpirer, 2022-11-11:
Same request for update for 22US or even information how to build it myself