TheCrypt0/yi-hack-v4 #20: 4FCN problem

BabyMaple, 2019-03-17:
Hi all 

I try yi-hack-v3 but the firmware y20 cannot flash 
(cannot ssh / web)

Any guys try b4 4FCN  ?

someone said change name from y20 to y23 will run 
but i am not sure it this can work or not 

hope yi-hack-v4 can solve

TheCrypt0, 2019-03-17:
Hi @BabyMaple 
I'm planning to support this version too. However it looks like I can't find it anywhere in Italy, I need to have it on hands to make a dump of the firmware.

BabyMaple, 2019-03-17:
> Hi @BabyMaple
> I'm planning to support this version too. However it looks like I can't find it anywhere in Italy, I need to have it on hands to make a dump of the firmware.

can teach me how dump the firmware so that i can share to u all

or i use DHL FEDEX pos to u also can if in the end i cant make it dump the firmware

BabyMaple, 2019-03-17:
btw just now i try change name from y20 to y23 and flash 

4FCN  really have flash the firmware , orange light blink around 30s 
after that ... totally no function T___T

BabyMaple, 2019-03-17:
i hv around 8 pcs of 4FCN ~ so sad all been lock in china ip ><

BabyMaple, 2019-03-17:
i just found the firmware at YI offical web

http://www.xiaoyi.com/home/firmware/download/
[image]

nicmorais, 2019-03-17:
@BabyMaple you can dump the firmware with an EEPROM reader, like this one:
[image]
Perhaps the files avaliable at YI's is enough to the developers to work. The OEM bootlog is also helpful.

BabyMaple, 2019-04-21:
> @BabyMaple you can dump the firmware with an EEPROM reader, like this one:
> [image]
> Perhaps the files avaliable at YI's is enough to the developers to work. The OEM bootlog is also helpful.
> 
> Edit: the [dd](http://priede.bf.lu.lv/ftp/pub/Faili/Riiki/CD-DVD/Create_ISO_from_USB_drive.html) tool can also be used to backup partitions.

hmmm.... i no really understand this 2 tool as i am a web programmer only 

now i have many 4FCN yi home carera 1080p 

or i can DHL FEDEX courrier to anyone for this project ? 

@TheCrypt0  @nicmorais  @aptonline  @gaggi

TheCrypt0, 2019-05-13:
@BabyMaple 

I've just bought a (Yi Home 1080p) 6FUS from Amazon IT and I'm working on supporting it in the firmware. Could you point me to an online store where to buy your version too?

BabyMaple, 2019-05-14:
I buy at this online store

https://detail.tmall.com/item.htm?id=550427778079&spm=a1z09.2.0.0.5d442e8d3Nf32L&_u=e3584bif7acf

china taobao Tmall

Jack
EMAIL : babymaple@gmail.com
Whatsapp : +60127181418
Skype : babymapleatgmaildotcom

hiol83, 2019-11-22:
Hi guys. Did anyone solve this problem?

sancho03, 2020-03-24:
Hi friends, I downloaded the firmware dump from the YI 1080p Home Camera (4FCN) CH341A programmer https://drive.google.com/file/d/1wlWeEt-EzEsV19TS4pWrFe6BylfowjTc/view?usp=sharing