TheCrypt0/yi-hack-v4 #21: homeassistant yi cam connection

tnTano, 2019-03-17:
hi to all...
i have one yi 1080p dome and two yi 1080p outdoor camera.

I installed the yi-hack-v4 firmware but i can't connect them to home assistant.... 
here the code i've got on HA:
`- platform: yi
  name: Corridoio
  host: !secret camera_ip_01
  password: !secret camera_pwd
  path: !secret ftp_url'
###
- platform: yi
  name: Ingresso
  host: !secret camera_ip_02
  password: !secret camera_pwd
  path: !secret ftp_url'
###
- platform: yi
  name: Cortile
  host: !secret camera_ip_03
  password: !secret camera_pwd
  path: !secret ftp_url`

the entities created succesfully but if i put it on the dashboard the entity are inactive

how can i resolve this problem?

thanks

Letherwin, 2019-03-18:
This is the configuration for my 1080p dome:

camera:
  - platform: ffmpeg
    name: Yi-Dome-1
    input: -an -i rtsp://192.168.1.250/ch0_0.h264

tnTano, 2019-03-18:
ok, is it working without using the YI component??

i'll try it...

thanks

TheCrypt0, 2019-04-10:
Hi @tnTano 
See this guide on how to configure the camera on Home Assistant: https://github.com/TheCrypt0/yi-hack-v4/wiki/Home-Assistant-configuration

Feel free to re-open if you have further issues.

closed 2019-04-10 by TheCrypt0

mitchross, 2019-06-09:
The wiki does not work.

codeclinic, 2019-08-08:
@TheCrypt0 The link you provided above just point to a blank page editor. Any chance of it being restored? Thanks.

TheCrypt0, 2019-08-09:
@mitchross @codeclinic 

Someone must have edited the page. Fixed!