TheCrypt0/yi-hack-v4 #24: [Feature] Iptables support

xvolte, 2019-03-18:
Hello,

I would love to have an iptables within the camera, to ensure nothing goes out of my local network, ie : on the chinese cloud, or on internet.

Would it be possible to have such feature ?

thanks in advance,

fbrinker, 2019-03-20:
Wouldn't it be a better option to place the device in a closed network or block it with an external (router) firewall?

TheCrypt0, 2019-04-10:
@xvolte 
I just tried to compile iptables for the camera to see what the size footprint would be. 
It works but the size of the executable is too big to be included in the firmware, let me know if you need some help in compiling it.

The last working version I managed to compile is `1.6.2`.

Cheers

closed 2019-04-10 by TheCrypt0