TheCrypt0/yi-hack-v4 #26: Yi Home YR-113-IR support

erykosky, 2019-03-20:
Firmwares yi-hack-v3 and v4 doesn't support this version. I only found one that support YR-113 do you can look into this firmware

https://github.com/npetrovski/yi-hack-easy

and adopt it to yours?

andrixxxx, 2019-03-21:
check your serial number should start with 17CN / 27US / 47US

erykosky, 2019-03-21:
My serials starts with 22EU

xvolte, 2019-03-21:
same for me.
i guess we need to wait for @TheCrypt0  answer!

TheCrypt0, 2019-03-24:
Hi @erykosky,
I had a look at that firmware. It seems like your version runs on an Hi3518eV100 (this hack currently supports the V200 only). However I think that an implementation may be possible.

Unfortunately @npetrovski did not publish the procedure on how to compile its firmware and pack it.

Plus, I'd need to have one camera of that particular model on hands.

Temporarily closing because of lack of documentation about the other firmware. If you manage to get in contact with him and get the all required steps to create it feel free to re-open the issue.

closed 2019-03-24 by TheCrypt0