TheCrypt0/yi-hack-v4 #27: Yi Dome 1080p :: Web interface ?

PatrickR68, 2019-03-22:
At first, many thanks for what you are doing for us.

I have updated the firmware with yi-hack-v4 0.03 
it works fine, I get the "Status/About" page shown when I enter the IP of the camera in the browser (FireFox). 

The access via app to the camera is also working. 

Could I also access to the camera to rotate or more via web browser like firefox/chrome or other to move the camera or to see a live picture ?

Many thanks,
Patrick

indubwestep, 2019-03-22:
I agree, there doesn't seem to be any configuration option. How would I hook it up to homeassistant? How do I find the rtsp url or ftp into the camera? I'm confused... This seems like a great hack, just wish I knew how to use it.

TheCrypt0, 2019-03-24:
Hi @PatrickR68,
Those are all good ideas, thanks!

The pan/tilt feature from the web interface is going to be developed really soon (I've already made some demos and they work nicely). However the live picture from the browser will be a little bit tricky since modern browsers don't support direct RTSP streaming.

Remember that you can follow the development roadmap on Trello: [https://trello.com/b/EtuK8577/yi-hack-v4](https://trello.com/b/EtuK8577/yi-hack-v4)

PatrickR68, 2019-03-24:
Hi Davide,

many thanks for your answer and your good job.

Sorry, I thought it is already implemented. 
No problem, I know by my own, how difficult it is to work on all projects.

If I or maybe all the other's which are interested in your hack, can help you, 
let it us know.

Patrick

monkeydust, 2019-08-02:
Hi cyrpto - when will pan/tilt be released, will the interface be exposed so home assistant could control it? Keep up the awesome work!

GauntPT, 2019-09-07:
Would be really nice to move the camera.
Is there any command line (internal yi firmware low level hardware commands) to move it manually?

tnTano, 2019-11-06:
it's a great nice to have... with this feat i can finally use the indoor cam...