TheCrypt0/yi-hack-v4 #29: YI 1080p Cloud Camera support

projectsun2, 2019-03-22:
I know this camera has no Hi Chipset, but a RTS3903N. Is there a possibility to support it? Or even a possibility to start a rtsp server?
[https://www.amazon.de/FLOUREON-%C3%9Cberwachungskamera-Netzwerkkamera-Sicherheitskamera-Bewegungserkennung/dp/B07M766DJ4/ref=bmx_1/257-5832789-0931813?_encoding=UTF8&pd_rd_i=B07M766DJ4&pd_rd_r=657770d7-4cec-11e9-978a-714b33377841&pd_rd_w=Mu8AR&pd_rd_wg=MAWF9&pf_rd_p=5ebe6468-481b-4dfb-b5f6-485a181d413b&pf_rd_r=K472NDY3D1CQCFC7F7VF&psc=1&refRID=K472NDY3D1CQCFC7F7VF]

TheCrypt0, 2019-03-24:
Hi @projectsun2 
I was looking at that camera two weeks ago on Ebay (they are pretty cheap ~25$) and curious to see if an implementation would be possible. However I currently don't have much time to support other cameras.

I don't exclude that it may be possible in the future.

Closing for now.

closed 2019-03-24 by TheCrypt0

jruth85, 2019-04-25:
Hey projectsun2, just so you know that camera has an telnet open on it. The username is root, no password. I checked for UART connections on the board first and didn't find any so I thought I was out of luck till I saw that the telnet was open. I'll let you know if I find out anything else.

projectsun2, 2019-04-25:
That would be nice. I know, telnet is open. I was logged in already and searched for config files and services.
But I have no idea how to compile a firmware.

TheCrypt0, 2019-04-25:
Hey,
I was planning to support that camera too.

It would be great to know the output of `cat /proc/cpuinfo` and get the
filesystem dumped to the SD card.

Cheers

jruth85, 2019-04-25:
Here's cpuinfo and version

/proc # cat version
Linux version 3.10.27 (zhengqianbin@XY-201) (gcc version 4.8.5 20150209 (prerelease) (Realtek RSDK-4.8.5p1 Build 2521) ) #15 PREEMPT Mon May 28 14:33:43 CST 2018

/proc # cat cpuinfo
system type             : RLX Linux for IPCam Platform
machine                 : Unknown
processor               : 0
cpu model               : Taroko V0.2  FPU V0.1
BogoMIPS                : 497.66
tlb_entries             : 64
mips16 implemented      : yes

I'll try to dump the filesystem later.

projectsun2, 2019-04-29:
Hi, is the filesystem already dumped? How can I accomplish this, to help you?

jruth85, 2019-04-30:
Okay. I just did a dd of each mtdblock (0-7). Hopefully that works for you. If not, let me know.

[dump.zip](https://github.com/TheCrypt0/yi-hack-v4/files/3129666/dump.zip)

GirosPita, 2019-05-24:
Hope this is still alive. I got another brand, probably the same inside. There are many the same shipset inside.
I just wonder if it differs with others, image attached.
[image]

RyanEwen, 2019-09-20:
Just got a CACAGOO brand camera that uses YI IoT app:
https://www.amazon.ca/gp/product/B07VLFQNYQ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I get the exact same output as @jruth85 posted above. Telnet is open, root with no password. Hoping for RTSP!

holazit, 2019-09-28:
I have bought a Mibao 1080P camera (D600) that uses Yi IoT app and I have the same output of "cat cpuinfo" as @jruth85 and @RyanEwen. I hope that one day we can activate RTSP on these cameras.

highlandr, 2019-10-01:
Same here... telnet, root ...

derex33, 2019-10-29:
Hello, I'm still a beginner andi  try to integrate my Mibao D500 with RTS3903N chip in ioBroker.  I tried to connect the Cam via VLC player. On ispyconnect i found the URL from Mibao D100.  Unfortunately it did not work with that: rtsp://root@192.168.0.36/live/ch0

I tried a lot of variants, but everything was unsuccessful. Can someone help me with the problem.

eliafino, 2019-11-20:
Hi.
I have another one Veroyi IPCAM, at boot search "home_r10m" in mSD for update...
A way to have RTSP?

samtap, 2019-12-14:
The Yi IoT Outdoor cam seems to be another clone with exactly the same specs. If the Realtek RSDK-4.8.5p1 s available somewhere maybe it would be worth another look. I'm a bit disappointed about the video quality in the Yi IoT app though. 
@TheCrypt0 Are you still interested in adding support for more cams or is this project more or less dead?

foulek57, 2020-01-25:
Hello, any news for this cam ?

movinDotz, 2020-02-16:
Any news on this camera model?
If you need any info/ image dump on this device let me know

foulek57, 2020-02-16:
Hi, i try to hack it with a friend, but is not easy, we have install a general busybox, but we have no all the tool that we need, but we have open the port 80 and the telenet port to have easy acces to the linux terminal.

We try to stream the tmp.mp4.tmp but we have no succces, we have no rstp or ffmpeg in this busybox, and do not know how to compile a busybox with this tools... 

If anyone have an idea to try to hack it together, please let me know.

hyperbit00, 2020-02-19:
> Hey, I was planning to support that camera too. It would be great to know the output of `cat /proc/cpuinfo` and get the filesystem dumped to the SD card. Cheers
> […](#)

any new informations about this cam :)?

DickyT, 2020-08-16:
> Hey projectsun2, just so you know that camera has telnet open on it. The username is root, no password. I checked for UART connections on the board first and didn't find any so I thought I was out of luck till I saw that the telnet was open. I'll let you know if I find out anything else.

Hey, I got a similar camera, but telnet does not work for me. Which port are you guys using for telnet?

crazygsm, 2020-08-18:
Hello @TheCrypt0 

I also have that camera and already dumped it all long time ago.

Only a folder /rom/ have R/W permissions where is store the wpa_supplicant besides other info, including a script where we can do some play to use it without the apps.

Anyway, replace all FW was the goal, let me know what you could need.

Was thinking if need to replace full flash content if needed.

crazygsm, 2020-08-19:
> Hey, I got a similar camera, but telnet does not work for me. Which port are you guys using for telnet?

In my case telnet was not active by default, you need to activate it using UART first.

Connect using uart

go to folder /rom
check if you have a script with name "setup.sh" and append there the line bellow.

telnetd

Save changes and reboot.

lordhaer, 2021-01-19:
Hi, I also have  yi cloud (Floureon).
There is some progress?

cjj25, 2021-07-14:
> Hey, I got a similar camera, but telnet does not work for me. Which port are you guys using for telnet?

If you're still interested in enabling telnet, see [my repo](https://github.com/cjj25/yi-hack-telnet-root) that has a trick.

cjj25, 2021-08-05:
> Hi, I also have yi cloud (Floureon).
> There is some progress?

I've now published a simple version that exposes an RTSPServer for RTS3903N based cameras here:
 https://github.com/cjj25/Yi-RTS3903N-RTSPServer (very early development but it works)

lordhaer, 2021-08-06:
> > Hi, I also have yi cloud (Floureon).
> > There is some progress?
> 
> I've now published a simple version that exposes an RTSPServer for RTS3903N based cameras here:
> https://github.com/cjj25/Yi-RTS3903N-RTSPServer (very early development but it works)

Hello, I have this Floureon YI camera, (not to be confused with YI from Xiaomi):
https://www.amazon.com/gp/product/B07KWWW82N/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I unzip the content of Yi-RTS3903N-RTSPServerV01.tar.gz
on an 8gb sd card formatted in fat32, at the root.

With the camera off, I insert the SD card and then when I turn it on, the rtsp service does not work. That could be happening?

The firmware version from the YI mobile application is:
7.0.00.90A_201904301632 Could that be the cause?

From the mobile application you cannot find another more current version.

How can I put the firmware version 7.1.00.25A_202002271051?

Thank you very much, you have done a great job!

cjj25, 2021-08-06:
It looks like you've done everything correctly and thanks for the feedback.

What's happening exactly? Does it still connect to the network?

I've tested this with `7.1.00.25A_202002271051` and `7.1.00.17A_201909271014`.

Are you 100% sure it's a Realtek based camera?

Your camera features an RJ45 port so it might be worth trying the hack without WiFi first (did you follow the wpa_supplicant instruction?) and see if you have a telnet port open. 

It's also worth attempting to backup the firmware first, follow the instructions on my other repo here: 
https://github.com/cjj25/yi-hack-telnet-root 

Once you're in telnet, run `cat /proc/version` and you should see something like
 `Linux version 3.10.27 (zhangzhidan@XY-201) (gcc version 4.8.5 20150209 (prerelease) (Realtek RSDK-4.8.5p1 Build 2521) ) #29 PREEMPT Wed May 22 14:16:12 CST 2019` - if yours doesn't say Realtek then one of the other hacks will probably work for you.

I'd be interested to see a copy of the backup too if you can transfer via WeTransfer or something similar.

Open an issue on the RTSPServer repo (just so we don't pollute this one)

cjj25, 2021-08-06:
> How can I put the firmware version 7.1.00.25A_202002271051?
> 
> Thank you very much, you have done a great job!

I've found the problem, the init script is different in your firmware. I'll post a fix shortly.
I'll update the issue here: https://github.com/cjj25/Yi-RTS3903N-RTSPServer/issues/1

Thank you @jruth85 for sharing your firmware dump!

ruinidavide, 2021-08-08:
Hi thank you for the job. 

I've an yi iot camera (mibao camera) with 7.1.00.11A_201905311440 firmware.

I unzip the content of Yi-RTS3903N-RTSPServerV01.tar.gz
on an 8gb sd card formatted in fat32, at the root.

With the camera off, I insert the SD card and then when I turn it on, the rtsp service does not work and the camera appear all time offline.

cjj25, 2021-08-08:
Have you modified wpa_supplicant.conf in /sdcard/Factory/ with your WiFi credentials? 

@ruinidavide  pls reply and open a ticket on the other repo.