TheCrypt0/yi-hack-v4 #32: Yi Home 1080P , orange light keep blinking during firmware install

justin8277, 2019-03-27:
Good day,
I bought a Yi Home 1080P (6FCNY0EHXE) 4 units yesterday, configured it based on the guide line as describe accordingly to yi-hack-v4 0.1.1. unfortunately is not working according.
based on my observation, after power on with sd card attached (software: rootfs_y20 and home_y20) the unit response with continuously orange blinking light until i power off..

hope to receive any valuable advise.

Thank you
Justin
-

mitchross, 2019-04-24:
I've had the same issue and had to return my camera due to brick, I think the Yi home 1080 cannot take the Yi Hack firmware.

danielandastro, 2019-05-31:
It happened to me, so I erased the card, put the files back, and it started up ok