TheCrypt0/yi-hack-v4 #33: Can't set the timezone

nicmorais, 2019-03-28:
In the 0.1.0 version, the timezone could be easily set by putting a TZ file into the /etc/ directory. But in the latest version (0.1.1) it's not working anymore.
So, how can I set the timezone in the latest version? Thank you.

nicmorais, 2019-04-07:
So, this is how you can set the timezone in 0.1.0 version:
1. Connect to the camera via SSH
2. Get your timezone string from [here](https://oldwiki.archive.openwrt.org/doc/uci/system)
3. Run `echo yourTZstring > /etc/TZ` e.g. if your timezone is America/New_York, then, you should run `echo EST5EDT,M3.2.0,M11.1.0 > /etc/TZ`
4. Run `chmod 0444 /etc/TZ`, so the app will not change the timezone of the device.
5. Reboot the camera.

TheCrypt0, 2019-04-10:
@nicmorais 
Thanks for the report. The version `0.1.1` features a minor change on how the date/time is set after boot.

Did you try to enable NTP in the Web Interface and check if the issue is still there?

Cheers

nicmorais, 2019-04-11:
@TheCrypt0 I remember enabling and disabling the NTP in the web interface, and both options did not solve that issue... But I'm not sure.
I'll be testing it in 2 or 3 days, then, I'll be reporting here.
Thank you! :)

nicmorais, 2019-04-13:
Solved! @
Flashed 0.1.1, set the timezone (in the /etc/TZ file) and rebooted. Works like a charm, even with NTPD client on or off.
Now, we need to write a tutorial on how to set the timezone in the README.md file.

TheCrypt0, 2019-04-13:
@nicmorais

Great news! I've given you write access to the Wiki, feel free to add a new page on how to set the timezone if you can. 

Cheers

closed 2019-04-13 by nicmorais