TheCrypt0/yi-hack-v4 #34: Url to get a snapshot

flashbackpa, 2019-04-05:
Hi,
would it be possible to have a url to retrieve snapshots from the cam, like for example:

http://ip.cam/image.jpg

Thank you.

TheCrypt0, 2019-04-10:
@flashbackpa 
This is currently not possible since the jpeg encoder is controlled by Xiaomi proprietary processes.

However it might be possible in the future if we discover that the snapshots are mapped in the camera RAM.

Closing for now.

closed 2019-04-10 by TheCrypt0

sr-1989, 2020-04-18:
Hi,
@TheCrypt0 Nice Job ! great work thanks for that ! 
@flashbackpa , i was looking for the same thing, url to take snapshot for domotic use, when i saw the command in expert mode in chome i saw that this URL is called :
http://192.168.1.X:8080/cgi-bin/snapshot.sh?res=high&base64=yes
i modified the base64 setting to no and seems to work ! 
http://192.168.1.X:8080/cgi-bin/snapshot.sh?res=high&base64=no
i've a snaphshot but not in jpg format...

flashbackpa, 2020-04-18:
> Hi,
> @TheCrypt0 Nice Job ! great work thanks for that !
> @flashbackpa , i was looking for the same thing, url to take snapshot for domotic use, when i saw the command in expert mode in chome i saw [...]

Hi @sr-1989, just tried, without port 8080 ( I guess you configured your cam with this port) but all I have is a 404 Not Found error...any idea ?

sr-1989, 2020-04-19:
> > Hi,
> > @TheCrypt0 Nice Job ! great work thanks for that !
> > @flashbackpa , i was looking for the same thing, url to take snapshot for domotic use, when i saw the command in expert mode in chome i saw that this URL is called :
> > http://192.168.1.X:8080/cgi-bin/snapshot.sh?res=high&base64=yes
> > i modified the base64 setting to no and seems to work !
> > http://192.168.1.X:8080/cgi-bin/snapshot.sh?res=high&base64=no
> > i've a snaphshot but not in jpg format...
> 
> Hi @sr-1989, just tried, with and without port 8080 but all I have is a 404 Not Found error...any idea ?

Hi,
maybe we have different FW version...
i've :

FW version : 0.2.5
Base Version | 4.5.0.0C_201910080934
Model Suffix | y203c
(i've maybe flash a fork version oh this hack)
on my html view i've a sanpshot tab and when i click on it you can have a snapshot view.
if you click on "F12" on chrome and refresh the page you could see, on the network tab,the different pages called. For me i saw a url with "snapshot.sh".

flashbackpa, 2020-04-19:
@sr-1989 thanks for those infos, could you please tell me where did you get FW 0.2.5 ?
In https://github.com/TheCrypt0/yi-hack-v4 I find 0.2.0 that seems the last version...
Could you please extract via ssh the snapshot.sh script and send it to me ?
That would be a great help, thank you in advance !
Roberto.

sr-1989, 2020-04-19:
> @sr-1989 thanks for those infos, could you please tell me where did you get FW 0.2.5 ?
> In https://github.com/TheCrypt0/yi-hack-v4 I find 0.2.0 that seems the last version...
> Could you please extract [...]

@flashbackpa , have a look here ;)
https://github.com/roleoroleo/yi-hack-MStar

flashbackpa, 2020-04-19:
@sr-1989 thank you, I didn't knew about the fork, I wonder what happened to @TheCrypt0 ?

sr-1989, 2020-04-19:
> @sr-1989 thank you, I didn't knew about the fork, I wonder what happened to @TheCrypt0 ?

be careful, be sure you have the good HW, as it's write : 

> This firmware is completely based on the work done by TheCrypt0 https://github.com/TheCrypt0/yi-hack-v4 It's a clone made for Yi cameras based on MStar platform.
But you can't choose between TheCrypt0 and MStar: if your cam is based on Hi3518e Chipset then you have to use TheCrypt0, if your cam is based on MStar Infinity Chipset then you have to use MStar.