TheCrypt0/yi-hack-v4 #35: ProxyChains-ng configration is gone on 0.1.1

fanlessfan, 2019-04-08:
the ProxyChains-ng configration is gone. how to config it in the latest version?

thx

[image]

TheCrypt0, 2019-04-10:
Hi,
The proxychains configuration can be found on the following path:

`/home/yi-hack-v4/etc/proxychains.conf`

It's currently not configurable from the web interface.

nicmorais, 2019-04-17:
1. Connect via SSH or Telnet (the default password for user root is blank)
2. Run `vi /home/yi-hack-v4/etc/proxychains.conf`
3. Edit the  [Proxylist] section and put in there your list of proxies. e.g. 
```
[Proxylist]
socks5  192.168.67.78  1080   login     secret
http    192.168.89.3   8080   username  hidden
socks4  192.168.1.49   1080
http    192.168.39.93  8080	
```
4. Exit vi: press Esc, type `:wq` and then press Enter
* [This tutorial](https://linuxhint.com/proxychains-tutorial/) can help you out.

joesster, 2019-07-16:
hi can i enter the proxychains and which directory should i go? bit confuse here.