TheCrypt0/yi-hack-v4 #36: Bricked y20 1080p

bgroins, 2019-04-10:
I seemed to have bricked a Yi Home 1080P y20 after installing this. The lights flash as normal, and I was even able to flash the recovery. However, when it powers up the orange light stays on for about 15 seconds, the camera makes an audible click sound, then goes dark. Anyone else have this issue before? I'm nervous about trying to flash more cameras.

bgroins, 2019-04-10:
Fixed my own issue. I was using a 2 GB SD card, and switched to a 32 GB. Was able to flash the recovery image and I'm guessing the hack will work as well. You might want to change "preferably of capacity 16gb or less" on the instructions, because there seems to be an issue with cards that are too small.

bgroins, 2019-04-10:
Verified... hack works fine with the larger card.

bgroins, 2019-04-10:
OK I'm an idiot. I had my status light set to off.

Vormillion, 2019-04-10:
HI @bgroins - may I know where did you buy this camera? I'm trying to buy compatible model but description is not clear at all.

TheCrypt0, 2019-04-10:
@bgroins 
Great to know. If you have any other problems feel free to open an issue! 

Closing.

closed 2019-04-10 by TheCrypt0