TheCrypt0/yi-hack-v4 #37: This camera can only be used within china

xristoforos7, 2019-04-10:
Hello, i flash the firmware succesfully but when i tried to pair it with the yi home china apk it says that "This camera can only be used within china"
[image]

nicmorais, 2019-04-18:
#### [Which Smartphone App to use?](https://github.com/shadow-1/yi-hack-v3#which-smartphone-app-to-use)
"The Chinese version of the camera will only work with the Chinese version of the app.

**Android users**, download the Chinese version of the Yi Home app. It is available from the Mi Store. [Click here](http://app.mi.com/details?id=com.ants360.yicamera)

* Alternatively, an APK file which can be installed manually is available from [here](https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr) under the directory 'Yi Home - Android'

**iPhone users** will need to change their App Store to the Chinese App Store and install Chinese version of the Yi Home app. [Link](https://itunes.apple.com/cn/app/xiao-yi-zhi-neng-she-xiang-ji/id931168730)

This firmware includes Proxychains-ng. This allows communication between the camera and Xiaomi server to be routed through a proxy server. If Proxychains-ng is configured with a proxy server from Mainland China, the camera will no longer be region locked."

xristoforos7, 2019-04-18:
Thanks for the reply, i saw the proxychains-ng file but it is empty. How can i configure it and what should i do after?

nicmorais, 2019-04-18:
Hey @xristoforos7 
You can set Proxychains by running `vi /home/yi-hack-v4/etc/proxychains.conf`. In the [Proxylist] section, put the proxies you want your camera to use. e.g.
```
[Proxylist]
socks5  192.168.67.78  1080   login     secret
http    192.168.89.3   8080   username  hidden
socks4  192.168.1.49   1080
http    192.168.39.93  8080	
```
Take a look at [this tutorial](https://linuxhint.com/proxychains-tutorial/).
Then, exit vi:

* Press Esc, type `:wq` and press Enter.

And reboot the camera:
`reboot`

xristoforos7, 2019-04-19:
I did all of this but still get this message, maybe all the proxies that i found are not working. Do you have a working proxies or a list to put it in?

nicmorais, 2019-04-19:
Take a look at [this list](http://spys.one/free-proxy-list/CN/).
Check if the protocols (socks5, socks4, etc) are correct and in lowercase.
My camera is a 27US, so I don't know how to properly work around the region lock...
What about the chinese app? Have you tried with it?
Regards.

xristoforos7, 2019-04-19:
I have the Yi 1080p Home CN V2.2. I have tested all this proxies no one works for me maybe, i have the tested with a lot of versions of the Chinese version of the Yi Home app and still nothing. I think the problem is with the proxies cause i had the yi-hack-v3 and when i test the proxies they all give me 3 red crosses

Rohja, 2019-04-23:
I've been wondering why Proxychain is the prefered solution to bypass the restriction.

@xristoforos7 : I only have a Yi Home 17CN so it might be different for other models, but in my case, the geo-restriction check is done client side and can be bypassed by [doing this change to the `cloudAPI` script](https://github.com/Rohja/yi-hack-v4/commit/08b984cb01402a5f62e5d74b187c4ac414bc70b5).
If this fix works, you will still need the chinese app (I use Android and no need of a proxy neither).

xristoforos7, 2019-04-23:
@Rohja Where can i find the cloudAPI script and how can i configure it

nicmorais, 2019-04-23:
@xristoforos7 cloudAPI is at **/home/yi-hack-v4/script/cloudAPI**
You can edit it with vi: `vi /home/yi-hack-v4/script/cloudAPI`
* Tip: if you are pressing keys on your keyboard but still can't edit the file, press the key `a`

To exit vi press Esc, type `:wq` (write and quit) and press enter.

xristoforos7, 2019-04-23:
I configure it but again it tells "This camera can only be used within china"

Rohja, 2019-04-23:
@xristoforos7 : Can you show us the full content of you `cloudAPI` script after the edit ?

nicmorais, 2019-04-23:
@Rohja @xristoforos7 
`cat /home/yi-hack-v4/script/cloudAPI`

xristoforos7, 2019-04-23:
[image]

Rohja, 2019-04-23:
@xristoforos7 You edit seems good. Did you restart the camera after the edit ? It's required to apply the changes.

xristoforos7, 2019-04-23:
@Rohja Yes i did a reboot..

xristoforos7, 2019-04-23:
Which apk you use, can you send me the link to download it. I have tried a lot of versions..

Rohja, 2019-04-23:
Whan I can guess is that my fix might only works on 17CN and/or the newer models are not hitting anymore `https://api.xiaoyi.com/v4/ipc/check_did` and a new endpoint is used. 

You can try to use install this version of `cloudAPI` in order to check what parameters are forwareded to `cloudAPI_real`:

```
#!/bin/sh

CONF_FILE="etc/system.conf"

if [ -d "/usr/yi-hack-v4" ]; then
        YI_HACK_PREFIX="/usr/yi-hack-v4"
elif [ -d "/home/yi-hack-v4" ]; then
        YI_HACK_PREFIX="/home/yi-hack-v4"
fi

get_config()
{
	key=$1
	grep $1 $YI_HACK_V4_PREFIX/$CONF_FILE | cut -d "=" -f2
}

# Log parameters forwared to cloudAPI_real
echo "$@" >> /tmp/cloudAPI_real_parameters.log
                
if [ "$4" == "https://api.xiaoyi.com/v4/ipc/check_did" ]; then
	echo '{"allow":true,"code":"20000"}'
elif [[ $(get_config PROXYCHAINSNG) == "yes" ]] ; then
	$YI_HACK_V4_PREFIX/bin/proxychains4 ./cloudAPI_real "$@"
else
	./cloudAPI_real "$@"
fi
```

You can then reboot the camera, wait a minute, do `cat /tmp/cloudAPI_real_parameters.log` and show us the result.

xristoforos7, 2019-04-23:
It says: 
/home/yi-hack-v4 # cat /tmp/cloudAPI_real_parameters.log
cat: can't open '/tmp/cloudAPI_real_parameters.log': No such file or directory

TheCrypt0, 2019-04-23:
Hey,
The problem might be on these lines:

```
if [ -d "/usr/yi-hack-v4" ]; then
        YI_HACK_PREFIX="/usr/yi-hack-v4"
elif [ -d "/home/yi-hack-v4" ]; then
        YI_HACK_PREFIX="/home/yi-hack-v4"
fi
```

They should be:

```
if [ -d "/usr/yi-hack-v4" ]; then
        YI_HACK_V4_PREFIX="/usr/yi-hack-v4"
elif [ -d "/home/yi-hack-v4" ]; then
        YI_HACK_V4_PREFIX="/home/yi-hack-v4"
fi
```

The file that is used is `/home/app/cloudAPI`, not the one in the `script/` folder. It gets replaced on the first boot after the firmware installation.

Let me know if this helps.

Cheers

xristoforos7, 2019-04-23:
@TheCrypt0 
I tried and now i get again the message ""This camera can only be used within china"

Rohja, 2019-04-24:
@xristoforos7 : With @TheCrypt0 's fix you should be able to make Proxychains works.

xristoforos7, 2019-04-24:
I have make a proxy list from a lot of sites but i think noone works

xristoforos7, 2019-04-24:
@Rohja I have make a proxy list from a lot of sites but i think noone works. I think that if i find a working CN proxy my camera will work again because when i had the yi hack v3 and the web interface cofiguration found automatically proxies my camera works perfectly, but now it doesn't find automatically proxies and my camera stopped working

naveednajam, 2019-04-24:
@xristoforos7 
path for the log file is incorrect it should be /tmp/sd/ instead of /tmp/

use the below line and try again if you can manage to get the logs. it is vital to get the logs to verify the URL 
# Log parameters forwared to cloudAPI_real
echo "$@" >> /tmp/sd/cloudAPI_real_parameters.log

xristoforos7, 2019-04-25:
@naveednajam The same thing: /home/yi-hack-v4 # cat /tmp/sd/cloudAPI_real_parameters.log
cat: can't open '/tmp/sd/cloudAPI_real_parameters.log': No such file or directory

TheCrypt0, 2019-04-25:
Make sure that you're editing the `/home/app/cloudAPI` file.

xristoforos7, 2019-04-25:
I edit this one: vi /home/yi-hack-v4/script/cloudAPI

naveednajam, 2019-04-25:
there is another copy of CloudAPI at /home/app/CloudAPI can you edit this
file and check again.
also show the contents of this file with cat /home/app/CloudAPI

xristoforos7, 2019-04-25:
@TheCrypt0 @naveednajam  I edit the:  vi /home/app/cloudAPI and it says pairing successful but it stucks in the app here and after a while says to check the internet connection
[image]

naveednajam, 2019-04-25:
@xristoforos7 try connecting on the same wifi and check

xristoforos7, 2019-04-25:
@naveednajam On the same wifi i am connected

naveednajam, 2019-04-25:
@xristoforos7  I am using iPhone to connect to my camera and have no issue. I did not try with Android

xristoforos7, 2019-04-25:
I tried a lot of versions....still the same. I will never make this camera work again...

TheCrypt0, 2019-04-25:
Mmmh, at this point I'm guessing that Xiaomi might have blocked the access even if you are using a proxy from the camera. 

Would it be possible to try to use a china proxy on your phone too to see the results?

Unfortunately I don't have a chinese version of the camera to test on.

Rohja, 2019-05-01:
From my experience, there's no restriction for the Chinese app on Android and iPhone.
It's possible to check the stream from anywhere in the world, even if the camera is meant to be only used in China.

ozeraser, 2019-05-02:
> @TheCrypt0 @naveednajam I edit the: vi /home/app/cloudAPI and it says pairing successful but it stucks in the app here and after a while says to check the internet connection

All, so i had the same issue.  Chinese only, turned on proxy-ng, added multiple proxies, no avail camera kept saying "china only".

I changed the script to get the debug log as per @Rohja above. [https://github.com/TheCrypt0/yi-hack-v4/issues/37#issuecomment-485868655](https://github.com/TheCrypt0/yi-hack-v4/issues/37#issuecomment-485868655)

The camera paired successfully after a reboot with this script. But when I went to view the camera I saw the same 25% issue and check internet connection. I replaced the CloudAPI script with the one from yi-hack-v4, and I still saw the 25% issue, I clicked settings cog in the CN App and it says Camera ON/Off and it was Off. I toggled that and the camera started working.

I got the logs from the CloudAPI for @Rohja, they are as follows (I removed any ids macs etc.);
`
-c 136 -url http://api.xiaoyi.com/v2/ipc/sync_time
-c 311 -url https://api.xiaoyi.com/v4/ipc/check_did -uid TNPCHND-197009-PWEDF -keySec !!REMOVED!!
-c 139 -keySec !!REMOVED!! -url https://api.xiaoyi.com/v4/ipc/reset -uid !!REMOVED!! -version 3.0.0.0D_201809111054 -mac !!REMOVED!!
-c 414 -url http://log.xiaoyi.com/info.gif?function=p2ptrack&version=3.0.0.0D_201809111054&did=!!REMOVED!! -filename /tmp/debug_p2p.txt
-c 138 -url https://api.xiaoyi.com/v4/ipc/on_line -key !!REMOVED!! -keySec !!REMOVED!! -uid !!REMOVED!! -version 3.0.0.0D_201809111054 -ssid !!REMOVED!! -mac !!REMOVED!! -ip !!REMOVED!! -signal_quality 100 -packetloss 0 -p2pconnect 0 -p2pconnect_success 0 -tfstat 10000
-c 137 -key !!REMOVED!! -keySec cAXN0QCgv4AHOr8p -url https://api.xiaoyi.com/v4/ipc/qr_bind -uid !!REMOVED!!
-c 142 -url https://api.xiaoyi.com/v4/ipc/deviceinfo -uid !!REMOVED!! -keySec !!REMOVED!!
-c 141 -url https://api.xiaoyi.com/v4/tnp/on_line -keySec !!REMOVED!! -uid !!REMOVED!! -version 3.0.0.0D_201809111054
-c 136 -url http://miwifi.com/api-third-party/service/datacenter/get_storage_info
`

naveednajam, 2019-05-02:
@xristoforos7 As per logs from @ozeraser At the second line we can see the URL is still same for 1080p camera and 17CN 720p camera. So it is confirmed issue is related to ON/OFF switch inside the app. 
-c 311 -url https://api.xiaoyi.com/v4/ipc/check_did

Rohja, 2019-05-02:
If someone having this issue use Android and knows how to pull logs using `adb` or Android Studio, we might be able to get some insight about what's happening on the apk side.
Mobile developers sometimes print interesting informations in the logs even when running the released version (and not dev).

phoxy84, 2019-05-21:
Great jobI I got it to work as well. First time I use my Chines Cam in The Netherlands! Thanks all!

TheCrypt0, 2019-05-21:
Fixed in this commit: https://github.com/TheCrypt0/yi-hack-v4/commit/74cfb7f3d3d7af6a2845fd8edb307e121bb49dbf

Closing

closed 2019-05-21 by TheCrypt0

kssssa, 2019-06-16:
still the same problem on 18cn 1080p

halnovemila, 2019-08-18:
How about this?

[SOLVED] Xiaomi Xiao Yi HOME CN12 – This camera can only be used within China. https://controsensi.blogspot.com/2019/08/solved-xiaomi-xiao-yi-home-cn12-this.html

hcrayfish, 2019-10-19:
> Fixed in this commit: [74cfb7f](https://github.com/TheCrypt0/yi-hack-v4/commit/74cfb7f3d3d7af6a2845fd8edb307e121bb49dbf)
> 
> Closing

can i ask how to solve the 25% issue?