TheCrypt0/yi-hack-v4 #41: Yi Outdoor Camera 1080p not working with 0.1.0 and 0.1.1

gdoumen, 2019-04-13:
I have a Yi Outdoor 1080p. 

I was able to successfully flash yi-hack-v3.

Then I tried to install the latest version of yi-hack-v4 (0.1.1), but the camera was not booting ( yellow lights were not blinking / just remained out)
Same happens with yi-hack-v4 v0.1.0.

When flashing yi-hack-v4 0.0.3, the camera was working again ( could be flashed & could be paired with the app )

TheCrypt0, 2019-04-14:
Hi @gdoumen 
Thanks for reporting your problem. After installing the 0.1.1 version what was the state of the yellow light?

Cheers

gdoumen, 2019-04-14:
Hi @TheCrypt0,
the light just remained dark 
Cheers

TheCrypt0, 2019-04-14:
Thanks for the info.

Try this:
1. Install a working version of the firmware.
2. Pair with the app.
3. Flash the 0.1.1 version (without resetting the camera).
4. Try to connect to it via SSH or by checking the Web Interface.

Let me know the results.

gdoumen, 2019-04-14:
Stupid question: How to I flash without resetting? 

Do I just overwrite the rootfs_h30 and home_h30, without changing any of the other files, or is there  specific procedure (or even OTA)?

TheCrypt0, 2019-04-14:
Exactly, the configuration of the network connection as well as some app settings are in a separate partition that's not affected when you update the fw.

Just put the rootfs_h30 and home_h30 files on the SD card and reboot the camera.

Let me know!

gdoumen, 2019-04-14:
That was working fine. Now app and web page are working. 
Thanks!

Thus, to summarize:

1) Deleting all files on SD. copying rootfs_h30 and home_h30, rebooting 
   => error  (works for 0.0.3, but not for 0.10 /0.1.1
2) Keeping files on SD. copying rootfs_h30 and home_h30, rebooting  
  => success

gdoumen, 2019-04-14:
One more observation:
- When flashing without reset, the www directory was still showing the old content (0.0.3), i.e. configuration menu and reboot camera menu was missing.

I manually fixed it, but overwriting it with the files from git repo, so this no issue for me right now, but I thought it might be worth to mention

TheCrypt0, 2019-04-25:
Thanks for explaining the process.

Closing

closed 2019-04-25 by TheCrypt0