TheCrypt0/yi-hack-v4 #43: [Feature]Access to full filesystem of SD card

gdoumen, 2019-04-20:
Maybe I just don't understand how the mp3 recordings are written to the file system, but when connecting via SSH or FTP, I am not able to access these files.

So I was wondering if it is not possible to make the FAT filesystem availbale ( i..e mount as directory similar to the /sdcard folder on mobile devices).

In my view this would provide two benefits:
- video recordings would be accessible, and could be even offered via <video> tag in HTML page
- Firmeare could be updated OTA (i.e. just overwrite the root_vXX and home_vXX files over fTP and reboot)

TheCrypt0, 2019-04-20:
Hi

The SD card is mounted on boot on `/tmp/sd`. Can you access it?

Cheers

gdoumen, 2019-04-20:
oh - I missed that. Thanks - closing ticket

closed 2019-04-20 by gdoumen