TheCrypt0/yi-hack-v4 #46: 12CN

SAOPP, 2019-04-24:
Guys!

Yesterday I trying to install 0.1.1 firmware to my 12cn with native firmware 1.8.5.1L_201510301024, but camera is not reacting to the sd-card and firmeware on it, just standart boot and start the write videos into sd-card.

I need just ask, maybe native firmware need to be updated before this hack?

I'm did try 0.1.0 also, and two new sd-cards. But results is not, camera is still wokrk on native firmware.

What I do wrong at the install-process?

nicmorais, 2019-04-25:
Hello
This hack does not support 12CN cameras at the moment. Perhaps this one does: [fritz-smh/yi-hack](https://github.com/fritz-smh/yi-hack)

TheCrypt0, 2019-04-25:
Apparently for each camera "user-friendly version", like "Yi Home" or "Yi
Dome" there are multiple sub-versions like 27US, 12CN, 47US and so on. I'm
trying to list all of those to catch the differences and support each one.
See this page for more info:
https://github.com/TheCrypt0/yi-hack-v4/wiki/Supported-Camera-Models

SAOPP, 2019-04-25:
Wow, apologize, I didn't see the 12cn model at the list of not supported models :( I think only old cameras like without 12cn number and only contains barcode/mac address at the back not supported... I also install at one of my old yi with mac at the back this fritz-smh/yi-hack and it's work very well.

Okay, so it's no vars for install latest of the hack to the 12cn model... Thanks guys!

TheCrypt0, 2019-04-26:
@SAOPP You can keep the issue open. Supporting that camera is planned and it's on the Trello Roadmap (https://trello.com/b/EtuK8577/yi-hack-v4 see the To-Do section). 

Finding one of those versions would be great.. developing without testing on camera can be pretty difficult.

SAOPP, 2019-04-26:
Okay, if it's needed it will be stay opened for future. Thanks.

Btw, just for the info for someone who trying to install this version of hack to old models, the fritz hack is work fine at this model too, with nightvision.

wilb, 2019-05-04:
@TheCrypt0 I might have a spare one I could post over if it helped. Would be nice to get it back at some point, but no major drama if it ends up bricked in a drawer :-)