TheCrypt0/yi-hack-v4 #47: Works with Clones?

Tw1st3dD3st1ny, 2019-04-26:
Hi just wondering if this works with clones like Viden 1080 dome cam 
https://www.amazon.co.uk/gp/product/B07PLNDT4R

Firmware 8.0.0046B_201811301813
As it works via the Yi IOT app and seems to link to there other websites I wanted to know before I test it out and likely brick the device as I do not know how to backup the firmware.
Cheers and look forward to your reply wish I bought the original as its cheaper but had to learn the hard way not keen on it messing with my privacy and sending it self off to China without my knowledge!

nicmorais, 2019-04-26:
We haven't tested any clones yet. Also, YI IoT is not the YI Home App. We need to know what CPU this device has, how much ram and rom, etc.

Tw1st3dD3st1ny, 2019-04-26:
Ok no probs I can give any info needed if I have a little guide I will try it with Yi Home as well looks like the email was sent from yitechnology.com so that made me think its on the same firmware as the 1080 dome cam. Cheers for your fast reply and great job on the project :)

nicmorais, 2019-04-27:
Thank you!
Could you please disassemble the camera and tell us what's the CPU, RAM and ROM chips?

Tw1st3dD3st1ny, 2019-04-27:
Not sure this is of any help it only seems to show the wifi and poe chips as I could not get any further inside without breaking the ribbons and stuff as it is a very tight fit :( https://ibb.co/tMPqLQC

Sorry I was hoping to get it apart more and wanted to even have a go at dusting off the arduino to make a serial connection but it looks so fragile and I cannot see where to dismantle further without breaking something.If there is any other hacks to telnet or something to show the hardware let me know and I will be happy to give that a try.

nicmorais, 2019-04-27:
Thanks for your try.
Telnet and ssh services are disabled by default in these cameras. First, we need to know what cpu, ram and rom chips this device has, but if you don't want to break it, so leave it the way it is.
Could you please take a screenshot of the files recorded in the sd card? I want to see the directories and filenames.

Tw1st3dD3st1ny, 2019-04-29:
Sorry for the delay here is a pic with the file structure https://ibb.co/WcjzmSH thanks again and hope it helps.

nicmorais, 2019-04-30:
Wow, nice
The filenames of this clone and the filenames of the YI cameras are the same!

Tw1st3dD3st1ny, 2019-05-01:
Cool I had a little bash today at trying getting it set up with there home app instead of the iot one.
Trouble was it would not get to the final stage of naming your camera after choosing any device from there list and if you tried to do it with the qr code it said you had to use the chinese app or something :(

So I reinstalled the iot app for the time being and thought maybe see if it can search for an update then I could sniff the traffic to hopefully reach the server url to get a firmware to reverse engineer but no dice it does not seem to even attempt to look one up and there is no changelog. Maybe someone with more experience could test it out or buy one to dismantle and work out the hardware?

Anyway thanks for all your help and best of luck with this project I will be installing it on a yi cam when I have some spare cash for a new one :)