TheCrypt0/yi-hack-v4 #49: Firmware not upgrading

ricveal, 2019-04-27:
Hi recently bought a Yi 1080P Home camera and followed the steps to upgrade the firmware but the led is not flashing yellow; It just completes the regular startup process.

I've check my current firmware version and it is 2.1.0.0E_201809191630 (Factory Firmware).

I'm using  [yi_hack_v4_v0.1.1__yi_home_1080p](https://github.com/TheCrypt0/yi-hack-v4/releases/download/v0.1.1/yi_hack_v4_v0.1.1__yi_home_1080p.zip) acording my camera version.

[yi-hack-v3](https://github.com/shadow-1/yi-hack-v3/issues/262) seems to be the same problem.

Could this firmware version be locking the upgrade to 3rd party firmwares?

benjimeistro, 2019-04-28:
I've just bought one of these too and I was experiencing similar issues, I have downloaded v3 and v4 and couldn't get the camera to flash from the SD, so I opened it up and have successfully connected to the serial port to grab some logs. 

I noticed the bootloader was looking for a file called home_y203c and rootfs_y203c (as well as various other files) so I renamed the v4 0.1.1 files from home_y20 to home_y203c and the rootfs_y20 to rootfs_y203c and gave it a flash.. :| It erased the memory and flashed the image successfully.. and then the log file went into overdrive complaining about jffs partitions and magic bitmasks (see attached logs for details).

Now i reckon i've bricked it, for the time being at least. Its looking like a new variant from what i have seen so far.. I will attach the logs to this post for in case they are useful to someone.
 
[boot-after-flash.log](https://github.com/TheCrypt0/yi-hack-v4/files/3125059/boot-after-flash.log)
[v3-flash.log](https://github.com/TheCrypt0/yi-hack-v4/files/3125089/v3-flash.log)
[recovery-v4-flash.log](https://github.com/TheCrypt0/yi-hack-v4/files/3125092/recovery-v4-flash.log)

It could be something i've not done correctly in the process also.. :)

version magic '3.4.35 mod_unload ARMv5 p2v8 ' should be '3.18.30 preempt mod_unload ARMv7 p2v8 '

Many Thanks, B

rxrasim, 2019-04-29:
I have the same problem. I tried to flash Dome 1080p.

rxrasim, 2019-04-30:
The problem is the new firmware. Need to revert to old firmware and try again. I did.

benjimeistro, 2019-04-30:
Hi, Are you suggesting that a downgrade will resolve this situation or that you have tried to downgrade the firmware? Thanks, B

rxrasim, 2019-04-30:
I downgraded the version of the official firmware and it solved the problem. 
On the last official firmware the device does not recognize the files (home_h20 for Dome 1080p), so you need to downgrade the firmware.
Sorry for my English))))

rxrasim, 2019-04-30:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/24964945596

benjimeistro, 2019-04-30:
Ahh great thanks, no worries. I just wanted to check. I'm going to try to downgrade this then. Thanks for your confirmation. I'll update either way :) Appreciated.

Sasso0101, 2019-04-30:
Hi, I'm new to Yi hack. How can I downgrade the firmware? Thanks!

TheCrypt0, 2019-04-30:
Hi everyone,
I've just bought a 5FCN camera to dump the firmware and start making some tests.

I'll let you know when arrives.

Scope666, 2019-05-03:
My Yi 1080p Home recently purchased from Amazon is showing version 4.5.0.0A_201902261502

Others in this thread have said they've successfully downgraded.  How did you do it?  What are you renaming the file to, which stock firmware file are you using?

Best I can tell benjimeistro is correct and it's not looking for **home_y20m** or even **home** anymore.

Thanks in advance... trying to figure out if this is unhackable in which case I'll return it.

benjimeistro, 2019-05-03:
Hi Scope, 

Don't upgrade this if it looks like Yi1080p, there's not a working hack at for this camera that I can work out.

Mines currently bricked in my drawer awaiting some logs pulling off of it.  It seems the dome can be downgraded successfully and then hacked. 

Hope this helps, B

Scope666, 2019-05-03:
Yeah, I think this is really a "Yi 1080p Home 3" but Amazon isn't calling it that or separating it out from the original SKU.

stephbairey, 2019-05-03:
Confirming that my Yi Home 1080p, sn 6FUS, will not flash. Purchased from amazon in late March.

TheCrypt0, 2019-05-03:
Here we go!

[image]

Scope666, 2019-05-03:
> Here we go!
> 
It's probably too early to tell but do you think this will be hackable?  I'm just wondering if I should return this and get another model that can accept the custom firmware.

TheCrypt0, 2019-05-04:
@benjimeistro Mine doesn't have any marking for the TX/RX serial pads, I'm currently probing them one at the time to see if I can get some output. On what pads did you connect?

@Scope666 It's too early to say that but I'm confident that it might be possible. Keep in mind that supporting a new camera with a completely new chipset requires some time, minimum >1-2 months.

Scope666, 2019-05-04:
> @Scope666 It's too early to say that but I'm confident that it might be possible. Keep in mind that supporting a new camera with a completely new chipset requires some time, minimum >1-2 months.

Thanks for the reply... I think I'll return this one, I originally bought it because I thought my original Yi Home was too old for your firmware, but when I opened it's 3rd party waterproof housing, I discovered I had a 27US, I now have v4 0.1.1 running on it, and I even managed to get MQTT working.   ;)

TheCrypt0, 2019-05-04:
Here we go! (pt.2) I'll dump the firmware to create a restore for the ones that bricked their camera.

[image]

TheCrypt0, 2019-05-04:
Here we go! (pt.3) I managed to find the MSTAR SDK and try to compile a simple Hello World... and it works!

[image]

benjimeistro, 2019-05-04:
Sorry i missed the message, glad you got it sorted! Well done on the progress so far!

TheCrypt0, 2019-05-06:
Just compiled Busybox for the camera and managed to get a working telnet connection. This machine is pretty powerful for its price.

[image]

lann, 2019-05-19:
Could you share which pads and settings you found for UART?

Fantoccini, 2019-05-21:
Having the same issue with `4FUS`, wondering is there anything I can help to solve this issue?

andy2301, 2019-05-24:
@TheCrypt0 great progress, congratulations! 
Do you mind sharing how you got access to the terminal? I tried TTL before on a 5FCN device (MSC313E chipset) but didn't succeed in getting the terminal access.
https://github.com/shadow-1/yi-hack-v3/issues/190#issuecomment-441584611

hustcalm, 2019-07-21:
> Here we go! (pt.3) I managed to find the MSTAR SDK and try to compile a simple Hello World... and it works!
> 
> [image]

@TheCrypt0, great progress! Can you tell us where you downloaded the SDK? I was trying to find the link, but no luck yet...

roleoroleo, 2019-08-18:
> 
> 
> Could you share which pads and settings you found for UART?

+1

BoscoLosco, 2019-08-24:
Hello, I'm new with yi cam, mine is a 6FUS, Yi 1080p Home. I would like to ask how the downgrade is performed, I tried to insert the home_y20 and rootfs_y20 files into a microSD file but the downgrade does not work, I read that the 6FUS hack is in progress but there is crypt in the change log that was added support for 6FUS. Do I still have to wait or is there a procedure to install it? I apologize for my English but I'm using google translator. Thank you

feiticeir0, 2019-09-30:
Just received my camera. Same starting serial number, firmware not flashed... Very interested on this ... No updates on the issue ?

zackishome, 2019-10-09:
@benjimeistro @TheCrypt0 could you share the picture to show where is the UART pads on this PCB?

zackishome, 2019-10-10:
@benjimeistro , thanks a lot , it is clear enough for me already, :), I have tried to probe , it just confuse me with my multimeter to see a lot of pads behave the same ...... Thanks for the detailed explanation and answer!

benjimeistro, 2019-10-10:
@zackishome No problem, you're welcome. Enjoy!

cseelhoff, 2019-11-29:
I am having a similar problem I think. I have a Yi Home 1080p 6FUS (YYS.2016) running 2.1.0.0E_201809191630. Is this firmware: https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases/ supposed to work with this camera? I can't seem to get any firmware loaded. Is that because it is not supported yet?

benjimeistro, 2019-11-29:
@cseelhoff just wanted to confirm, I successfully unbricked and flashed my 6FUS camera using the you have link provided above. 

I had to downgrade to version 2 of Roleos version and then upgrade to the later version. If this is failing then a downgrade to an earlier official firmware of the camera may work but I'm not sure where you can get the files. I think you can see the firmware version in the Yi app..

If you do manage to find it, then you might need to do a manual firmware flash using the reset button whilst plugging the camera into usb. 

There are some instructions available on one of the chinese yi camera websites.

Good luck!

cseelhoff, 2019-12-02:
I'm trying to go back to the original firmware (2.1.0.0A_201703071456home_y20m) from here: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/23321022068

I followed the steps here: https://www.yitechnology.com/Public/index/file/FirmwareUpdate.pdf to rename the file "home_y20m" and placed it in the root directory of the 16GB SD card (formatted as FAT32). Then I reinsert the card back into the camera. 

I plugged in my camera while holding the reset button to flash the firmware, but now it is just stuck on a blinking yellow light. Did I do these steps correctly?

cseelhoff, 2019-12-02:
@benjimeistro When you say you had to downgrade to version 2, do you mean you extracted this file: https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases/download/0.2.0/yi_home_1080p_6FUS_450_0.2.0.tgz to the root of your MicroSD, and then just plugged in power like normal, with only those two files present (home_y20 and rootfs_y20)?

benjimeistro, 2019-12-04:
@cseelhoff I formatted an SD card in fat32 then downloaded roleos version 1.2 and ensured there where no hidden files on the SD card.

 I then copied across the two files named home_y203 and rootfs_y203 front he hack to the root of the SD card and then inserted the sdcard into my camera and plugged in the usb power. The camera then re-partioned the corrupt filesystem on my bricked camera. 

Then I flashed the camera with version 0.2.0 to bring it up to a working version. Using the same method as before.

ricveal, 2019-12-04:
I've tested the last release available with my 1080 camera (SN 48US***) and the stock firmware 2.1.0.0E_201809191630 and everything works as expected now and I'm currently running this custom firmware. Thank you for your amazing work.

So, for my part, we can close this issue. Anyways, a lot of extra work and information has been added since I opened it and it seems like there are still some problems in specific camera versions.

I'm not sure if we should close this issue now and create a new one where we can follow these specific problems or just keep it open and continue the discussion here...

What do you think?

cseelhoff, 2019-12-05:
@benjimeistro when you say roleos version 1.2, do you mean this file? https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/archive/0.1.2.zip because I don't see see home_y203 or rootfs_y203 in the zip. 

When I look in this file: https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases/download/0.2.0/yi_home_1080p_6FUS_450_0.2.0.tgz , I see home_y203c and sys_y203c, but not home_y203 or rootfs_y203. Am I looking at the right files?

benjimeistro, 2019-12-05:
@cseelhoff Yeah you are they are the files you need to copy. @ricveal Yeah I am happy for this to be closed. I no longer see or have any issues with the firmware upgrade. Thanks!

ricveal, 2019-12-05:
Great! Thank you all!

closed 2019-12-05 by ricveal

chiquiff, 2019-12-20:
Hi all! I have a non-working 6FU 1080p Home Camera right next to me.

When I plug it in, a solid yellow light comes up, then a lens opening / closing noise, and finally a blinking yellow light. That in a never ending loop.

So I've seen @benjimeistro solution, but I don't understand how to install that `roleos 1.2`. Do I need to download this? https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/archive/0.1.2.zip

There's no home_y203 nor rootfs_y203 in there. :(

I've tried with this https://github.com/roleoroleo/yi-hack-6FUS_4.5.0/releases/download/0.2.0/yi_home_1080p_6FUS_450_0.2.0.tgz, but get stuck in the same yellow light loop.

Any ideas?

NielsFeenstra, 2020-01-05:
> Here we go! (pt.2) I'll dump the firmware to create a restore for the ones that bricked their camera.
> 
> [image]

Do you have the original firmware? So i can restore my bricked device? I'm having the same problems as @chiquiff 
After flashing a reboot loop with the following error;
random: dropbear urandom read with 103 bits of entropy available