TheCrypt0/yi-hack-v4 #51: Auto upload the recorded footage to the cloud.

MonSHut, 2019-05-01:
Hello TheCrypt0,

Please add the feature that camera can auto upload the recorded footage to the cloud like Google Drive, Dropbox or Youtube (at private mode).

Thank you.

hungpqz, 2019-06-23:
This feature may make use of curl and Google Drive API. It will be very nice feature

smsamadi, 2019-08-04:
please add it

mziach, 2019-09-15:
I'm looking for simpler solution (the best configurable via web interface), but will try this solution in the few days when I have time https://github.com/porunov/xiaomi_gdrive, maybe will work...

michael11123, 2019-10-02:
please add it to upload directly to gdrive,

michael11123, 2019-10-26:
For now, it just create for yi home 720p. I hope it will support for more devices.

Minims, 2021-04-29:
@michael11123 I have a Yi-Home 720p 27US. But I don't see how to setup google drive upload for this camera with yi-hack-v4. Can you point me the right direction ?

thirsty1975, 2021-04-29:
OK. Gib mir morgen bis Mitternacht.