TheCrypt0/yi-hack-v4 #52: Connecting to WiFi without a QR code

davidjb, 2019-05-01:
I'm keen & able to make a PR for this, but wanted to broach the topic first to get your thoughts, @TheCrypt0.  

When setting up a Yi camera and plugging in (temporary) wifi details to the app, I noticed that the QR code requires Internet access to be generated.  That's pretty concerning since your credentials are being sent into the ether, but also it's cumbersome as setup/changing details is a fiddly manual process (I like automation) but also dependent on their app.

I've got no interest in using their app so as an alternative, one can bypass the `dispatch` process and its QR reading by just supplying your own wifi config for `wpa_supplicant` and starting it on boot.  A very hacky solution is to pop this into `/tmp/sd/yi-hack-v4/startup.sh` (using `wpa_passphrase` to generate the `wpa_supplicant.conf`):

```bash
ifconfig wlan0 up
sleep 1
/home/base/tools/wpa_supplicant -c/tmp/sd/yi-hack-v4/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -iwlan0 -B &
/home/app/script/wifidhcp.sh &
```

and tada, wifi with no QR code needed.  Obviously, things could be a _lot_ better via init.d scripts or the like (and `dispatch` needs restarting to stop the annoying "Waiting to connect" blather) so hence me opening this issue first to ping you before I go off and implement a permanent solution. 

So, thoughts?

(Thanks for all the work you're doing on this btw!)

ozeraser, 2019-05-01:
Funnily enough i have been working on something similar this afternoon.

I have been reading the wiki on https://github.com/TheCrypt0/yi-hack-v4/wiki/Reverse-Engineering-the-QR-Code as I was not wanting to install the YI app and bugger about with region locking.

I also stumbled upon some open vulnerabilities in the camera's QR reading software. https://www.talosintelligence.com/vulnerability_reports/TALOS-2018-0572

You will notice that it states the application does not send the SSID or the password to the YI server or over the network, the image and encryption is created locally on your phone.

I currently have a hacky c# application that builds the encoded string that allows the camera to connect to WIFI, however as it cant verify me as I do not have a YI account or valid Bind ID (I'm using 0000001), on next power up it wont connect to wifi until I show it the QR, but your idea does mean i could bypass that and never need to use the YI app from initial purchase to using the device.

davidjb, 2019-05-01:
Interesting reading, thanks for the link on the vuln report.  Good to know the SSID/key aren't being shared but that's proof that the QR scanning is best avoided (and disabled).  In any case, the idea of having the camera 'just work' after flashing an SD card is what I'm aiming for.

nathankellenicki, 2019-08-14:
@davidjb So this is currently possible now? If I buy a new camera, put this firmware on the SD card, create a wpa_supplicant.conf with wpa_passphrase, put that and a startup script to move it to the right place on the SD card, I won't have any need for the app?

davidjb, 2019-08-14:
@nathankellenicki Yes and no.  You can get the camera to connect to WiFi without needing the app in the way I describe above and that'll allow you SSH access into the camera.  However, back when I was trying this out, the image capture internals wouldn't initialise until any old valid QR code was scanned (eg `viewd` and co wouldn't work until this happened).  So, in short, you don't need the app -- but for the camera to actually work with RTSP/image capture, you'd have to figure out how to get this initialisation to happen automatically.

This was a while ago so things might have changed with these hacks since then.  The image quality and poor IR in the Yi Outdoor camera was extremely disappointing so I've shifted focus to other cameras with far better quality imagery and a similar price point (that just work with RTSP etc).

noisemaker00, 2019-11-06:
Hi, I want to connect my YI with this hack firmware to my wlan without using any app.
I read the first post but I didn't understand the procedure. Where I need to put the file 'startup.sh'? I have no tmp folder on my microSD. Can you write a step-by-step tutorial?

Schnegg, 2020-02-24:
I am also interested in a WiFi connection solution via SD. And I also don't understand where to put the files. What do I have to do? Thank you very much.

Luke1962, 2020-07-07:
Hi, it would be very useful manually setting WiFi into some configuration file or via web interface, because I've discovered that Yi Home 3 (9FU...) does not accept Wifi connections that have not internet access!  
In my opinion this is a big limitation. I asked to Yi support more clarifications, but without any useful answer.

maki43, 2020-11-19:
To resurrect this...I have tried your instructions and placed the code in startup.sh and generated the wpa_supplicant.conf in the folder you mention but it doesn't change the Wi-Fi connection on reboot.  I have already hacked it and connected via the QR code but i wanted to change it manually rather than using the app again.  Any help appreciated.