TheCrypt0/yi-hack-v4 #55: How to bypass "Can only be used in China" region lock?

fulwater, 2019-05-04:
Hi guys. I brought my camera back from China to Asia recently and I've been looking for a way to use it. v3 didnt work because of the dead proxy list. I am not tech inclined so I'm asking for a bit more simplified help if possible.

My unit is a 1080 Dome Camera (Black).

Regards

nicmorais, 2019-05-04:
Take a look at [this issue](https://github.com/TheCrypt0/yi-hack-v4/issues/37)
From what I know, you must set Proxychains-ng and download the [chinese YI home app](https://github.com/shadow-1/yi-hack-v3#which-smartphone-app-to-use)
My camera is not chinese, so I don't know how to properly bypass the region lock...

TheCrypt0, 2019-05-08:
Duplicate issue, please refer to the issue that @nicmorais linked.

Closing.

closed 2019-05-08 by TheCrypt0