TheCrypt0/yi-hack-v4 #58: A few questions about the hack

maisun, 2019-05-08:
1) Does it support Yi home 12CN device?
2) Is there a way to turn on/off camera from telnet/ssh?
3) Can the camera still be paired with Mi Home app?

Thank you in advance!

nicmorais, 2019-05-11:
1. No. This is a very old model of the Yi Home camera (and might use a slightly different chipset)
2. Yes. Check the web interface
3. Yes.

Also, take a look at our [wiki](https://github.com/TheCrypt0/yi-hack-v4/wiki)

nicmorais, 2019-05-12:
1. No. This is a very old model of the Yi Home camera (might use a slightly different chipset)
2. You can only reboot it via ssh and telnet. But, it's possible to stop all the cloud services 
3. Yes

Also, take a look into our [wiki](https://github.com/TheCrypt0/yi-hack-v4/wiki).

boscorelly, 2019-05-14:
With the hack, is it possible to record to a nas ?

nicmorais, 2019-05-15:
@boscorelly Yes! Take a look at this: [yi-hack-dl](https://gist.github.com/nicmorais/cf490ee2cf069312fc6d7431b3faec10)