TheCrypt0/yi-hack-v4 #59: How i can help with auto upload the recorded footage to the cloud function?

GramJr, 2019-05-10:


nicmorais, 2019-05-12:
What do you mean by 'cloud'? AWS? Google Drive?
If so, Take a look into projects like [gdrive](https://github.com/gdrive-org/gdrive). From what I know, you must compact it and crossmake it.

smsamadi, 2019-08-04:
please add this feature

thoms3n, 2019-08-15:
Dropbox or OneDrive Support would be great

nicmorais, 2019-09-08:
Hello guys
I made [this script](https://gist.github.com/nicmorais/7c28265526ad25cb0386fbfdbe990252) which uploads the files in the camera's SD card to [Backblaze's B2 Cloud Storage](https://www.backblaze.com/b2/cloud-storage.html) (that's pretty cheap cloud storage service) 
I know, this script needs some optimization, but for now it is working.
Also, it was made to run in a "normal" computer or a Raspberry Pi: it's not going to run in any camera.

nicmorais, 2019-09-16:
@thoms3n Good news: the uploader script for Dropbox is ready! See [YI-Hack-Tools/YI-Hack-Dropbox](https://github.com/Yi-Hack-Tools/YI-Hack-Dropbox). Again, this is not going to run in any camera, but we are close to achieve it: Crypto said that curl support will be added soon. Stay tuned.

denven, 2023-02-02:
- For Microsoft OneDrive files uploader, check  [yihack-onedrive-uploader](https://github.com/denven/yihack-onedrive-uploader)
- For Google Drive files uploader, check [yi-hack-MStar-gdrive](https://github.com/roleoroleo/yi-hack-MStar.gdrive)