TheCrypt0/yi-hack-v4 #60: RTSP username/password

cirolaferrara, 2019-05-10:
Please add this feature

magidel, 2019-05-25:
ok, right ... and not just for RTSP!
Is there a way to access the cam's web server via LAN (Yi Home 720 in my case) using credentials?
Anyone accessing my network (WiFi or Ethernet) can access the webcam panel via IP... potentially dangerous, can't you find?

ovals12, 2019-05-27:
I would also vote for authentication if it is possible to make there.

colas31, 2019-07-24:
Adding my  vote too. 
Having a full security on the both RTSP / website too.

Thanks :)

darianf01, 2020-01-27:
take my vote as well! it's really important to have auth to the web interface and rtsp stream

vmendivil, 2020-06-23:
I was looking how to set a user/password to RTSP and website and found out there is an open issue with this request. Hope can be added!!!!

AdiyZd, 2025-05-04:
password default rtsp?