TheCrypt0/yi-hack-v4 #61: Yi Hack Not Flashing 1080p Dome

erikarruda, 2019-05-13:
I've been trying to flash my yi dome 1080p 65US (black color) with the hack but i don't know why it is not working, i've successfuly been able to flash a downgrade to 1.9.2C firmware version but when i try to flash with the hack it does not work, the camera starts normaly, i think that i was looking for the home_h20m file but the hack is named as home_h20, when i tried to change it to home_20mthe camera didn't started the yellow light came on and the blue light flashed some times but it didn't work, how sould i proceed to make it happen:

TheCrypt0, 2019-05-13:
Try to format the SD card directly by using the Yi App. Then place the home_h20 and rootfs_h20 on the SD card.

erikarruda, 2019-05-13:
thank you man, it did the trick, now just waiting for the RSTP licence.

closed 2019-05-13 by erikarruda