TheCrypt0/yi-hack-v4 #64: yi 1080p home camera

ale953, 2019-05-17:
I have bought a yi 1080p home camera 6FUS, and I want to know if it's possible to hack this model of the camera

nicmorais, 2019-05-18:
According to the wiki, the support for the new 5FCN, 5FUS, 6FUS cameras (based on a different chipset) and older cameras is on the to-do list.

ale953, 2019-05-21:
so i have to wait?

nicmorais, 2019-05-23:
@ale953 Yes...

snicker, 2019-05-30:
had some of my older cameras get cooked, the new ones coming from amazon seem to be these 6FUS. Willing to offer some help where I can to get these working

ale953, 2019-06-12:
sorry @snicker I can't understand what you had written

hanyunfan, 2019-06-20:
I'd like to know once it supports 6FUS. Thanks.

Tomasu, 2019-07-01:
On a related note, does this project support the 24US model? If not, is there any way I can help work on support? Or is there another known working alternative firmware for it? (I've tried the old now unsupported yi-hack-v2 that sorta works, but the video can be unstable).

jrotaetxe, 2019-07-19:
Just to encourage this.. Another frustrated 6FUS

hustcalm, 2019-07-21:
Any updates on the 6FUS camera? @TheCrypt0 Do we have a rough date when it will be available?
I'm more than happy to do any help (development, testing, etc.)!

nrrdzilla, 2019-07-29:
So I just got one of these from Amazon, and it's a 6FUS, naturally, which explains why none of the features described on the yi-hack-v4 project page worked. If there's a testing branch, I'd also be willing to help with debugging.

TheCrypt0, 2019-07-29:
Hi everyone, big news for you.
Another developer has just joined the project and will help to bring support for the 6FUS, this will speed up the whole roadmap and allow faster development!

Welcome abroad @shxdow!
He's a super smart guy, has a lot of experience in reverse engineering, low level C/C++ and security research.

More updates coming soon!

maelboutin, 2019-07-29:
That is good news,

I'm a developper too so in case any help would be required for testing, i could try :)

Prigent-HighConnexion, 2019-08-06:
Yeah good news. My arms are up to give you the power  :)

hustcalm, 2019-08-06:
Definitely good news!!!

I'm also a developer with experience on embedded Linux development, willing to do any help with Development/Testing:-)

nrrdzilla, 2019-08-06:
Hi all,

I'm not that much of a seasoned developer, but I will do what I can, and
have my 6FUS camera aimed at a couple of bunnies for us to use in testing
new firmware images. There *will* be lots of motion activation whenever
they're not sleeping, and possibly some non-PG content, as they are bunnies
and try to assert dominance 😅

haha,
M

olegpopadko, 2019-08-16:
Hello all.

I've just received my new YI 1080p Home BFUS. As I understand this camera isn't supported - BF > 6F. Am I right?

And if 6F support should be soon, BF is not so soon 😺

ghiglie, 2019-08-24:
+1 for 24US!

SurfingSpider, 2019-09-16:
Following

Also just discovered I have a 6FUS Camera. Doh. 

Can I recommend that you make it a little clearer on the "Supported Cameras" page that you have to look at the first four digits of the QR Code on the rear of the camera (or DID number on the rear of the box) to find the variant. I spent a while knowing I had a 1080p home but no clue which variant of it I had.

As always thanks for your efforts on this project. Not a complaint just a suggestion.

steef84, 2019-09-28:
Prerelease 6fus here https://github.com/roleoroleo/yi-hack-6FUS_4.5.0

ghiglie, 2019-09-28:
Oh damn, I returned the cam!
I didn't realize how much work it's behind these firmwares. Good job!

nrrdzilla, 2019-09-28:
Nice. I'll try this later today, or tomorrow.

I'll report on it, after I've had a chance to play around.

Thanks so much,
M

ffix, 2019-09-30:
Do 5FUS has the same hardware as 6FUS or it requires another port?

websterjazz, 2019-11-12:
I followed the instructions and added the two files to the root for a 32gb sandisk card. Now my YI 1080P  Home cameras orange light is continuously on. Even after removing the card its on.. And it doesn't get reset. What to do?

VikIborra, 2020-03-04:
+1 for 24US ?

nrrdzilla, 2020-04-22:
Things are working well with my 6FUS -- I just wanted to say thank you to all who contributed to this effort!

PS I've done a bit of testing to put the aforementioned bunnycam online but, while it works fine on my WLAN/LAN, I haven't had time to satisfactorily config my firewall for exposure to the wilds of the internet.

lucasilvestri, 2020-07-15:
+1 for 24US!

Symbol84, 2021-03-13:
Please! 24US

zoral566, 2021-04-03:
one more 24US please :}