TheCrypt0/yi-hack-v4 #65: Camera doesn't reconnect to MQTT broker if the broker goes offline temporarily

alexanv1, 2019-05-20:
Whenever, the MQTT broker goes offline temporarily, camera doesn't reconnect to the broker when it comes back online - i.e. motion events are no longer received by the broker.

Solution is to reboot the camera. I run my own broker and restart it relatively recently when changing config, etc so this becomes very tedious with several cameras.

It would be great if camera could reconnect to the broker same as other MQTT clients do.

SeydX, 2019-05-26:
That would be great!

davidgurr, 2019-05-29:
I've used code like this before to cover this (note, I'm not up to speed with GitHub and have no time to learn, so don't ask me to pull or push something):

```
	if ((err = mosquitto_loop(mosq, -1, 1)) != MOSQ_ERR_SUCCESS) {
		fprintf(stderr, "Mosq loop error: %u\n", err);
		fflush(stderr);

	if ((err == MOSQ_ERR_CONN_LOST) || (err == MOSQ_ERR_KEEPALIVE) || (err == MOSQ_ERR_NO_CONN)){
		wait_for_network();
		mosquitto_reconnect(mosq);
	}
```
Together with:

```
#include <sys/socket.h>
#include <linux/wireless.h>
#include <sys/ioctl.h>

void wait_for_network()
{
	int sockfd = socket(AF_INET, SOCK_DGRAM, 0);
	struct ifreq ifr = {.ifr_name = "wlan0"};

	if (debug)
		fprintf(stderr, "Waiting for network\n");

	while (true) {
		if (ioctl(sockfd, SIOCGIFADDR, &ifr) == 0) {
			ioctl(sockfd, SIOCGIFFLAGS, &ifr);
			if(debug) fprintf(stderr, "Flags: %x\n", ifr.ifr_flags);
			if (ifr.ifr_flags & IFF_RUNNING)
				break;
		}

		sleep(15);
	}

	if (debug)
		fprintf(stderr, "Network connection now up\n");
}
```

nicolasgomollon, 2019-06-01:
Any update on this? I’m running into the same issue with my setup.

lance36, 2019-06-15:
@TheCrypt0 this should be fixable by using `mosquitto_loop_start();` which automatically handles reconnection instead of `mosquitto_loop();`

SeydX, 2019-06-15:
@TheCrypt0 

Can you look into this? Its really annoying to restart the cam every time :(
Im using the mqtt function for my Homebridge plugin

TheCrypt0, 2019-06-15:
Thank you guys for the report of the issue I don't use MQTT myself, testing is the only way to find such issues.

I'm currently a little bit busy for my main job, if someone could send a PR and testi it on the [mqttv4 repo](https://github.com/TheCrypt0/mqttv4/) I'll merge it.

manny199, 2019-07-13:
@lance36 where can I insert mosquitto_loop_start(); which automatically handles reconnection instead of mosquitto_loop()? I do not understand how to use your istructions

tinysnake, 2019-11-15:
I use motioneye to send mqtt the other day, and I uses `mosquitto_pub` to send mqtt message.
I think the way `mosquitto_pub` send message is quite a good solution.
It disconnect to the server right after the message been sent.
If we do that, then we don't have to worry about the stability of long time connection anymore.
But there's a downside though, connecting to the broker every single time may be slower, if the connection time is less than 1 second, then I think it's not a problem anymore.