TheCrypt0/yi-hack-v4 #66: Adding RTSP stream to Homebridge

WEZANGO, 2019-05-20:
Did anybody managed to add RTSP stream from Yi Hack V4 to homebridge? I can watch the stream through VLC but in HomeKit it's only black screen without any video. 

I am using regular sapmle config provided at homebridge ffmpeg plugin wiki

`{
  "platform": "Camera-ffmpeg",
  "cameras": [
    {
      "name": "Camera Name",
      "videoConfig": {
        "source": "-re -i rtsp://myfancy_rtsp_stream",
        "stillImageSource": "-i rtsp://myfancy_rtsp_stream",
        "maxStreams": 2,
        "maxWidth": 1280,
        "maxHeight": 720,
        "maxFPS": 30
      }
    }
  ]
}`

SeydX, 2019-05-21:
yes, i have modified the homebridge-cameras-ffmpeg plugin and using also the mqtt service from this hack

xvolte, 2019-05-21:
@SeydX can you share what you did please?

WEZANGO, 2019-05-22:
> yes, i have modified the homebridge-cameras-ffmpeg plugin and using also the mqtt service from this hack

Are you using Raspberry Pi for that or something else?

SeydX, 2019-05-23:
@WEZANGO 

yea its a raspberry

@xvolte 

will publish a own plugin for this within next few days

SeydX, 2019-05-26:
v1 

https://github.com/SeydX/homebridge-camera-mqtt

WEZANGO, 2019-05-26:
So it is an Mqtt plugin. But how did you add camera itself? Actually I already managed to add my camera, but it connects in about 10 secınds and it has really low frame rate, freezes once in a few seconds. I would say it’s because raspberry is not really powerfull, but while watching camera, CPU is loaded only for 20%. Do you have the same issues?

SeydX, 2019-05-26:
@WEZANGO 

Its a ffmpeg (its also possible to use other codecs) plugin with mqtt support designed for the mqtt service which this hack offers

Im using this with a Intel NUC, with the example-config you can find on the main site of the plugin

Do you tried with ffmpeg?

SeydX, 2019-05-26:
changed from homebridge-camera-mqtt to homebridge-yi-camera

here some screenshots

https://github.com/SeydX/homebridge-yi-camera/tree/master/images

WEZANGO, 2019-05-27:
> 
> 
> changed from homebridge-camera-mqtt to homebridge-yi-camera
> 
> here some screenshots
> 
> https://github.com/SeydX/homebridge-yi-camera/tree/master/images

That is just great. Thank a lot!

SeydX, 2019-05-27:
Next version will add also a web interface for the camera to access the stream via http from overall

chriseckman, 2020-01-27:
@SeydX What changes did you need to make? I'm having similar issue with Shinobi.