TheCrypt0/yi-hack-v4 #67: Make MQTT independant from Yi app alerts

pszewello, 2019-05-21:
Hi,

I love the new MQTT and how well it integrates with Hassio, there's one problem though. This is the behavior I see, when I switch off motion detection in the Yi app the MQTT doesn't send anything (this makes sense, but it would be nice to have an independent implementation), when motion detection is on, the MQTT events are transmitted and also Yi app send notifications (alerts which are annoying), if I disable the alerts in Yi app by setting alerts time to custom then MQTT doesn't also send any messages... I would like to continue receiving mqtt notifications, but not the yi app alerts, do you think it's possible to develop that?

Cheers,
Piotr

TheCrypt0, 2019-05-21:
Hi @pszewello,

Thanks for your suggestion, unfortunately the current implementation catches the messages sent from the Xiaomi motion detection and forwards them via MQTT. One dirty fix would be to disable notifications from the Yi App.

By the way I think that's possible to stop those messages going thru the Cloud and allow them only via MQTT. I'll look into this in the future.

Cheers

lance36, 2019-06-14:
If you are using android, just disable notification permission on the app like I do.
They don't work reliably anyway...

closed 2019-06-15 by TheCrypt0