TheCrypt0/yi-hack-v4 #68: Time zone change

Veritas999, 2019-05-21:
Hi all

The names of the folders and recorded videos do not match the set time in the app settings.
How to change the firmware local time to the actual time?

danielandastro, 2019-05-23:
See #33

Veritas999, 2019-05-23:
Yes, but now rewind in Yi Home is not working and you cant select any record except the first/

nicmorais, 2019-05-24:
Perhaps you will have to rename all the directories in the sd card, to fix the hour they were recorded.
Also, I've had the same issue with rewind here, I don't know how to fix it.

Veritas999, 2019-05-24:
I tried to rename folders and files, even formatted a card.
Also ... My time zone is "MSK -4". In APP time is correct, but in firmware the time is displayed one hour less. Those to display my real time in TMZ, I need to put "...- 3".
But the problem is that for working rewind and viewing any file except the first is possible only with an empty TMZ file.

nicmorais, 2019-05-25:
Oh I think I figured it out.
1. Follow [these instructions](https://github.com/TheCrypt0/yi-hack-v4/wiki/Setting-the-Timezone) avaliable at the wiki 
2. In the YI app, go to camera settings and set the timezone there as UTC+00:00, Coordinated Universal Time
Done! Now, rewinding works again.

TheCrypt0, 2019-05-25:
Great news! The Wiki article should be updated too.

Veritas999, 2019-05-25:
Still not working for me (

nicmorais, 2019-05-26:
Rewinding is somehow working for me. My timezone is UTC-3, if I want to watch something recorded at 16:00, i have to put that white needle at 13:00. The needle "jumps" 3 hours forward.

Crescendo-BLYAT, 2019-06-18:
Timezone setting is weird:
GMT-1 is GMT+9 Japan.
GMT+1 is GMT+7 Vietnam.
GMT+2:30 is GMT+5:30 India.
GMT+8 is GMT+0 UK.
Find your timezone yourself :)

1onar, 2019-07-10:
Same issue here. I've set my timezone to +1. 
When i want to wach a File, like for example 16:00h, i have to select 17:00h.
Somebody got a answer to this problem or fixed it?