TheCrypt0/yi-hack-v4 #71: Can't access the camera after disabling cloud features

ovals12, 2019-05-24:
Hi,
I flashed yi-hack-v4 version 0.2.0 to my Yi Home 1080p camera. It basically works fine. I am waiting for rtsp enabling resources for which I already paid via discord but in the meantime I wanted to have some kind of access for viewing camera. I found out I am not able to view the camera anymore after disabling cloud features. Is this as designed? Even on wifi I am not able to connect to the camera via Yi Home android app.

So when cloud features are disabled only rtsp is usable?

TheCrypt0, 2019-05-24:
Hi @ovals12,

Yes, the Disable Cloud feature disables the App access. This way none of your video data is sent to the remote servers.

Cheers

ovals12, 2019-05-25:
Hi @TheCrypt0 ,
yes I understand it blocks data to be sent to a chinese cloud servers. But is there some way of viewing video from the camera locally (except of enabling rtsp)?

TheCrypt0, 2019-05-25:
Unfortunately not, there isn't. The stream is request by the App and then
sent directly to the servers.

The only way to get the stream locally is by using RTSP. (That's the reason
I wanted to make it work).

ovals12, 2019-05-27:
OK, clear. Thanks!

closed 2019-05-27 by ovals12