TheCrypt0/yi-hack-v4 #73: Compatible model

stopamarcin, 2019-05-28:
Is this compatible for Xiaomi cmsxj16a?
https://www.xiaomi.ua/ip-kamera-imi-xiaobai-smart-camera-ptz-edition-cmsxj16a/p22417/

nicmorais, 2019-05-30:
Hello
This model isn't supported yet.

GarretBlueMango, 2019-06-06:
I'm interested in an alternative firmware for this model too. Would like to help, but I'm completely new to 'hacking' these cameras. Can anyone point me to some reading material or tutorial to get started?

nicmorais, 2019-06-08:
@GarretBlueMango See our [wiki](https://github.com/TheCrypt0/yi-hack-v4/wiki) and join our [discord server](https://discordapp.com/invite/upPsFWZ)

andretg79, 2019-10-04:
Hi, is this model supported yet? :)

ronaldlokers, 2019-10-25:
Is there any progress on this model; if not how can I help?

ronaldlokers, 2019-11-20:
@TheCrypt0 Can you point me in the right direction on how to add support for this camera model myself? I might just need some pointers to get started.

nicmorais, 2019-11-24:
@ronaldlokers We need to know what SoC the device has, we need a bootlog, an eeprom dump, so we can get it started.

OUARZA, 2020-01-11:
Good morning all,
Do you have news about a hack for the CMSXJ16A model?
Thanks in advance.

jffz, 2020-02-16:
> @ronaldlokers We need to know what SoC the device has, we need a bootlog, an eeprom dump, so we can get it started.

I might be able to provide all of this but i am still wondering how to open it without breaking it first.
Also os there a guidé to do an eeprom dump ?

javierhurtado, 2020-08-17:
Hello, I'm interested to on this model :) CMSXJ16A
Thanks!