TheCrypt0/yi-hack-v4 #74: Update README for clearer understanding of supported Device/Model Numbers

hugheba, 2019-05-30:
Since there are several models that share the same name, update the README/Wiki with the working model numbers, i.e.

|Model|Not Supported|In Progress|Supported|
|---|---|---|---|
|Yi Home - 17CN| | |✅|
|Yi Home - 27US| | |✅|
|Yi Home - 47US| | |✅|
|Yi 1080p Home - 24US| ❌| | | 
|Yi 1080p Home - 5FCN| |🕐| |
|Yi 1080p Home - 5FUS| |🕐| |
|Yi 1080p Home - 6FUS| |🕐| |

Pinnhate, 2019-05-30:
And what about Yi Dome 720p - 63US?

nicmorais, 2019-05-30:
@Pinnhate it's supported

deditricahyono, 2019-06-01:
how is yi  dome 43 CN.. support? or not

nicmorais, 2019-06-01:
@deditricahyono I don't know.. 43US is supported, but 43CN isn't.

kssssa, 2019-06-03:
and how about 18CNY

galadril, 2019-06-13:
Its in the wiki:
https://github.com/TheCrypt0/yi-hack-v4/wiki/Supported-Camera-Models