TheCrypt0/yi-hack-v4 #78: Unable to connect outside of China

skywx, 2019-06-04:
Hi I am using Yi 1080p Dome (china version) camera. Attempting to connect it at region outside of China but still failed despite I have enabled proxychain via the web interface.

May I know how can I get the camera working?

Thanks.

kssssa, 2019-06-05:
i have the same problem with Yi 1080p home (18CN)

nicmorais, 2019-06-06:
See [this issue](https://github.com/TheCrypt0/yi-hack-v4/issues/37).

skywx, 2019-07-08:
So meaning I need to copy cloudAPI directory to microSD card, and place it along with the firmware in order to work? Is that correct?

ozeraser, 2019-07-28:
You need to add proxy servers, it may take a few goes as they are difficult and unreliable. See this https://github.com/TheCrypt0/yi-hack-v4/issues/37#issuecomment-484726170