TheCrypt0/yi-hack-v4 #79: Home Assistant Wiki Article Does Not Work

mitchross, 2019-06-09:
Followed the guide here https://github.com/TheCrypt0/yi-hack-v4/wiki/Home-Assistant-RTSP-configuration.

[image]
[image]

lance36, 2019-06-11:
to me it looks like something is wrong with your home-assistant installation, care to share some details on how is it installed?

can you also check the homea-assistant logfile? it could be either the ffmpeg or the stream component are not getting initialized correctly.
do you get a still image but no live-feed?

mitchross, 2019-06-12:
I used a raspberry pi 3 + this install  sudo curl -sL https://raw.githubusercontent.com/home-assistant/hassio-installer/master/hassio_install.sh | sudo bash -s -- -m raspberrypi3

These are two issues I see

https://github.com/home-assistant/home-assistant/issues/22840
https://github.com/home-assistant/home-assistant/issues/23561

closed 2019-06-14 by mitchross