TheCrypt0/yi-hack-v4 #84: Add documents in wiki for Synology Surveillance station

jokerigno, 2019-06-18:
HI,

I saw in the description that Synology Surveillance station is supported. But there's no description in how to do it.

Can someone please add it?

It seems that requires user, pass and port and there's no specification for it in the docs.

Thank you in advance

palla89, 2019-07-15:
Same question here!!

iamrob84, 2019-07-17:
Click 'IP Camera' on desktop.
Click 'Add' > 'Add Camera' in top left of window.
Select 'Complete Setup'.
Click 'Next'.

Enter the chosen 'Name' of Camera.
Select 'Brand' as 'User Defined'.
Select 'Type' as 'Streaming - RTSP'.
In 'Path' enter the URL for the SD or HD stream from the cameras webpage i.e. '192.168.0.70:554/ch0_1.h264'.
Click 'Test Connection'.

You may encounter an issue saying that it cannot communicate with the camera, in this instance try rebooting your Synology box and retrying the process.

Click 'Save'.

On success open the 'Edit' page for the camera.
On the 'Device Settings' panel, select the 'Advanced Tab'.
Set the following:
'Transport Protocol' : 'TCP'
'Keep-Alive method' : 'GET_PARAMETER'
'Streaming information source' : 'RTSP'

Hopefully that should get you up and running .

ranrinc, 2019-08-10:
@iamrob84 thanks for the detail documentation. However my connection after running in about 5 minute got cancel. And trying to edit the page after successfull setting wont allow me to save due to error messages. Unsure how to make it running

asimotran, 2020-11-02:
Here is my problem:
[image]

dimitriy-k, 2020-12-01:
what about ONVIF?

it can be added, but not activated:
[image]