TheCrypt0/yi-hack-v4 #89: RTSP does not work properly

baerni, 2019-06-24:
I have a problem with the RTSP. I checked in a few applications.  RTSP works very rarely, each time it either rejects the connection or shows a black screen. Several times a day a smooth image from the camera will appear. Often I have to reboot camera and sometimes the interface does not work and I have to disconnect the camera from the power supply. I have Yi Outdoor, yi-hack-v4 0.2.0, rtspv4__1.0.0

owenworley, 2019-07-23:
I see a similar issue. Yi outdoor 1080, latest hack b4 with license etc.

Rebooting the camera means  it will work for some hours, but eventually it will get to a state where it returns a blank stream on RTSP. Funnily enough, MQTT is still triggered for motion, but video is blank. Interesting.

Kilberz, 2019-07-29:
I also have this issue, the stream to VLC works fine but using some CCTV Monitoring Software I get a successful connection but a black video feed. Please help!

ranrinc, 2019-07-30:
hemm would like to know if this going to be a problem

owenworley, 2019-07-30:
> I also have this issue, the stream to VLC works fine but using some CCTV Monitoring Software I get a successful connection but a black video feed. Please help!

This is not the same as my problem. My issue is that rtsp does not work *at all* no vlc, no motioneye, no homebridge/homekit, no nothing.... whilst the native yi app *does* work and shows video.

I am assuming that there is some issue with the services that convert the video into rtsp format, or the rtsp streaming process itself.....

I have run 'ps aux' on camera before and after the bug happens, but I cannot see much difference - i could be wrong, buit it doesnt look like anything is crashing/not running at this point.

If someone could tell me what commands to run via SSH to kill and restart the services responsible for RTSP then I can give those a try to see if it is indeed an issue there, otherwise im not sure where to start diagnosing this. :(

AndreVallestero, 2019-08-06:
> This is not the same as my problem. My issue is that rtsp does not work at all no vlc, no motioneye, no homebridge/homekit, no nothing.... whilst the native yi app does work and shows video.

I'm also experiencing the same issue. Yi Home app works fine but the RTSP stream specifically does not work. I've tried accessing it via VLC with no luck. I have access to the RTSP page and it's enabled along with the HD stream but I'm not getting any stream at the supposed address.

TheCrypt0, 2019-08-10:
Thank you for the report. I don't have a Yi Outdoor cam but if you could give me the model number (it should be on the camera below a QR code) I'll try to get one from Amazon or eBay.

AndreVallestero, 2019-08-10:
Model: YHS.3017

owenworley, 2019-08-10:
> 
> 
> Thank you for the report. I don't have a Yi Outdoor cam but if you could give me the model number (it should be on the camera below a QR code) I'll try to get one from Amazon or eBay.

Yi Outdoor Camera 1080p
Model: YHS.3017

owenworley, 2019-08-10:
If there is anything I can do to get logs/try restarting process/etc please tell me, I am familiar with ssh/shell stuff

TheCrypt0, 2019-08-11:
> If there is anything I can do to get logs/try restarting process/etc please tell me, I am familiar with ssh/shell stuff

Sure, you can connect to the camera via SSH and manually start `viewd` (see `--help` for all the parameters) and then `rtspv4` (opening two shells would be preferred).

If something goes wrong you should see a warning or error message from one of those programs.

owenworley, 2019-08-11:
Will do this tonight and see if I can recreate the problem and capture the output

AndreVallestero, 2019-08-13:
@TheCrypt0 Here's the output of running `viewd` and `rtspv4`
[image]

TheCrypt0, 2019-08-16:
@AndreVallestero Thank you for the screenshots.

Just guessing, do you have a chinese model? It may happen that the encoder process stops if it detects that isn't in the right region.

AndreVallestero, 2019-08-16:
@TheCrypt0 I don't believe I have a Chinese model. It was purchased from Amazon and it works fine with the international version of the app.

GarretBlueMango, 2019-11-02:
I have similar problems with my Yi Outdoor Camera. If I log in via SSH I don't see viewd & rtspv4 running, if I start them manually (as deamon) they will work, but after a while the stream becomes unresponsive and if I log in via SSH same result no viewd & rtspv4 running. Any help or ideas?

derDeno, 2019-12-10:
same problem over here

Scope666, 2019-12-16:
Also having this problem.  RTSP  is available / unavailable in almost a predictable pattern.  Will try and capture some logs.

TheCrypt0, 2019-12-17:
Thank you for the report, I had to implement the feature to "prevent overlapping read" because sometimes the RTSP server was too slow to catch up with the frames provided by the camera. 

The logic behind that feature is:
- See how many frames in the buffer haven't been read yet.
- The buffer is a circular one, so if the number above is too high the writing head might overlap with the reading one and cause corrupted video.
```
R : Reading pointer
W : Writing pointer
- : Frames already read
+ : Frames to be read

Normal situation
| ======================== BUFFER ======================== |
| -----------------------------R+++W---------------------- |
| ======================================================== |

Overlapping read situation
| ======================== BUFFER ======================== |
| ++++++++++++++++++++++++++++++++++++++W-----R+++++++++++ |
| ======================================================== |
```

As you can see, the reading pointer is too slow it might get caught by the writing one and cause issues.

Anyway, it shouldn't happen this often.. could you please try with the SD stream and see if the issue persists?

Scope666, 2019-12-17:
Thanks for that very detailed explanation.  I have already tried the SD stream, as well is disabling EVERYTHING else on the camera(s) to free up as much CPU as possible.

I have an idea, since the RTSP server portion is having a hard time keeping up, what about having it process every other frame, perhaps with a switch option (-T or something)

Many NVR's record at 15 FPS, these Yi's seem to put out 20 FPS, so we have a bit to spare.

IMHO  A 10 FPS stable stream would be MUCH more useful than a 20 FPS stream that drops / restarts every 60 seconds or so.

owenworley, 2019-12-19:
> I have similar problems with my Yi Outdoor Camera. If I log in via SSH I don't see viewd & rtspv4 running, if I start them manually (as deamon) they will work, but after a while the stream becomes unresponsive and if I log in via SSH same result no viewd & rtspv4 running. Any help or ideas?

What commands are you using to restore the daemons? I’m looking to implement a temporary hack by scripting on interval/something Else like that.

alpat59, 2020-01-03:
I'm having the same your issues (even if the wifi signal is quite perfect). Currently RTSP stream is unusable. Is there some news about possible fixing? Thank you for your support

Scope666, 2020-01-03:
Just checked mine again.  Both are running SD streams only, no audio, cloud turned OFF, ftp off, etc:

[image]

For some reason the Outdoor drops WAY more often.

andyjenkinson, 2020-01-06:
Sorry for butting in; just having a browse through the project as thinking about using a couple of my Yi cameras for a project but haven't used it yet. I happened to read this issue and had some thoughts.

I believe I understand what the check is there to prevent, but unsure what it does when it finds this occurs, does it end the stream? I would expect to drop frames if processing is not keeping up with new data, especially for a streaming protocol. Can you just repeat the last frame and skip reading the "real" one until it catches up?

As for the root cause, I think /tmp/view is memory but does the output get written to disk (i.e. SD card)? If so writes to the card could be the cause if this blocks reading the next frame? I would be tempted to try to observe how reads from/writes to /tmp/view change over time (e.g. is the buffer being read smoothly but steadily falls behind, or do reads hang for long periods doing 'nothing' while writes continue?)

grungeguitarist, 2020-01-17:
I was thinking about buying the RTSP server functionality for all my cameras (esp. for my Yi outdoor cams). But if it's so unstable, I won't spend any money on it. Too bad the problem won't get fixed.

chriseckman, 2020-01-27:
[image]
I'm experiencing the same issue with a Yi Outdoor 1080p. The RTSP is working with VLC however it will not work on Shinobi. I just get a black screen. Anyone find a fix or work around for this?

chriseckman, 2020-01-27:
I was able to fix the issue and got it running. Turned out unraid will not write to an unassigned mounted NTFS drive.

barc593, 2020-04-27:
Im getting unstable rtsp, on yi dome 720p, where can i find this prevent overlapping feature?

krzada, 2021-03-21:
I am also experiencing such 'overlapping read' problem with YHS.3017. It is not a solution, but as a workaround I am using older version of rtspv4 - rtspv4__1.0.0, where everything works as expected. Hope it would help util fix is delivered.

krzada, 2021-03-22:
> I was able to fix the issue and got it running. Turned out unraid will not write to an unassigned mounted NTFS drive.

@chriseckman, could you please provide more details on your workaround ?

idanny76, 2022-01-14:
> I have a problem with the RTSP. I checked in a few applications. RTSP works very rarely, each time it either rejects the connection or shows a black screen. Several times a day a smooth image from [...]

Did you ever find a solution for this? I have an NVR with a couple of Yi Dome 1080p and after sometime the live feeds freeze and after a refresh it a black screen. Basically something is causing the cameras to loose connection. I am using the low quality RSTP feed on TCP on all cameras and the issue still persist. The only way is to either reboot the camera, reboot the NVR or re-enter the password on the NVR on each camera, which basically causes it to reconnect.

barc593, 2022-01-14:
It's a known issue, will not be solved. Sell the camera get a cheap v380
works much better

idanny76, 2022-01-23:
> It's a known issue, will not be solved. Sell the camera get a cheap v380 works much better
> […](#)

I was able to get the problem solved, thanks to RoleoRoleo from yi-hack Mstar. He was able to update or modify the rtsp server and it now automatically connects immediately when something may go wrong with the live feed.  Below is the link of the long chat troubleshooting the issue and how it was solved.

https://github.com/roleoroleo/yi-hack-MStar/issues/414

Scope666, 2022-01-23:
> > It's a known issue, will not be solved. Sell the camera get a cheap v380 works much better
> > […](#)
> 
> I was able to get the problem solved, thanks to RoleoRoleo from yi-hack Mstar. He was able to update or modify the rtsp server and it now automatically connects immediately when something may go wrong with the live feed. Below is the link of the long chat troubleshooting the issue and how it was solved.
> 
> [roleoroleo/yi-hack-MStar#414](https://github.com/roleoroleo/yi-hack-MStar/issues/414)

This alternate rtsp server doesn't run on the Yi Outdoor, I believe it has a different CPU.

```
/tmp/sd # ./rtsp_server_yi -r low -a alaw
Segmentation fault
```

idanny76, 2022-01-23:
> > > It's a known issue, will not be solved. Sell the camera get a cheap v380 works much better
> > > […](#)
> > 
> > 
> > I was able to get the problem solved, thanks to RoleoRoleo from yi-hack Mstar. He was able to update or modify the rtsp server and it now automatically connects immediately when something may go wrong with the live feed. Below is the link of the long chat troubleshooting the issue and how it was solved.
> > [roleoroleo/yi-hack-MStar#414](https://github.com/roleoroleo/yi-hack-MStar/issues/414)
> 
> This alternate rtsp server doesn't run on the Yi Outdoor, I believe it has a different CPU.
> 
> ```
> /tmp/sd # ./rtsp_server_yi -r low -a alaw
> Segmentation fault
> ```
> 
> Earlier binary he posted throws a different error:
> 
> ```
> /tmp/sd # ./rtsp_server_yi -r low -a alaw
> ./rtsp_server_yi: line 1: syntax error: unexpected word (expecting ")")
> ```

The fix is on the rtsp server file. RoleoRoleo gave me the files for the 720p yi  home and 1080p dome  indoor. Maybe crypto  can look into it. All I know my cameras are working fine now. I just wanted to share the info here just in case it could provide a fix also.