TheCrypt0/yi-hack-v4 #92: Yi Dome X Support

iltruma, 2019-06-27:
When you can add firmware for yi dome X?
it's the new model of yi dome 1080p

Thanks

f4t0, 2019-07-25:
hi. I need support for yi dome X too.
Thanks

gtramt1, 2019-07-30:
Hi, i would like to know if Dome X is international version of this one https://www.xiaoyi.com/homeCamera/homeCamera_h7/index.html

I have it and i can only use it via chinese VPN which are not reliable - flashing international firmware would fix this problem?

ykhandler, 2019-08-03:
@gtramt1, there are chinese version and also international version, the serial number begin *FCN in the bottom of the camera is chinese version *FUS is international version.
No, flashing international firmware wont fix the problem, the serial is checked on the camera and  server side to determine chinese or international version.

stream74, 2019-09-08:
HI, yes is it possible to have it for yi Dome X

fintmu, 2019-10-25:
I have Yi Dome X camera also and would very much like to add it to my Home Assistant environment. Any firmware coming soon for it? If any help needed for testing etc. please ask.

skeltonmfc, 2019-11-14:
So This hack doesn't work with the Dome X ?

fintmu, 2019-11-14:
As far as I know the answer is no. It's not a supported model. I have this camera and it's totally useless for me because I can not connect it to my Home Assistant environment.

skeltonmfc, 2019-11-14:
I would be happy to donate some money if we can get this going ?

Kugelfang666, 2019-11-29:
I would also be greatly interested in a solution for this model

pumabaer, 2020-01-02:
I would also be happy if this model is supported:)

davimas, 2020-04-23:
Hey, and now one year later - how is the status?

ma-karai, 2020-06-02:
any hope"?

skeltonmfc, 2020-06-29:
would really love this 2 !!!!!

winstarshl, 2020-09-11:
Any news for this cam?

fabertel, 2020-11-07:
me too interested .. what happens if we try to flash it anyway ? do we break it ?

antodeilippo, 2021-05-05:
I'm strongly interested, is there some way to hack to dome X and add it to home assistant ?

iltruma, 2021-05-05:
@antodeilippo Try the firmware of repository https://github.com/roleoroleo/yi-hack-MStar

iltruma, 2021-05-05:
Repository no longer mantaied for Yi Dome X

closed 2021-05-05 by iltruma