TheCrypt0/yi-hack-v4 #95: How do I restore original image?

stn4stn, 2019-06-30:
After trying the yi-hack v4 my camera is half bricked. 
I can connect to the camera via httpd and ssh, but the camera will not talk to the Yi application, and pushing the reset button does not do anything.

Does anyone know:

1) where I can find the original image
2) how I can re-flash it?

Thanks

awannet, 2019-10-27:
same issue here. The camera connected ok once. I changed the name on the webgui and reboot. Camera could not communicate with the app anymore. I tried to reset and add back, with or without the card to no avail.

How can we unbrick it? the section is blank.

I really want to go ahead and get the server add on and donate but so far the connection lasted less than 15 minutes before the camera self brick itself.

thanks

tried so far:
-reboot camera by webgui
-disconnect camera and reconnect camera  from power
-reset and rejoin. will reset but no rejoin. fails on pairing
-reset and rejoin with co flash card. will reset but no rejoin. fails on pairing

awannet, 2019-10-27:
how to unbrick 1080p home camera (works for me)
1) remove card
2) reset camera. as soon as it powers and you hear reset complete, waiting to connect, cut power
3) delete partition and format card on fat32 (maybe not required)
4) go to the next link, select the
5) https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/23494084746
6) verify that you have the proper shadow files, date is 2017, download the one for your cam
7) move them into the root of the card
8) insert the card, power the camera
9) add the camera with the yi app
10) once done, wait untill the camera does the update to the latest firmware
11) delete partition and format card on fat32 (maybe not required)
12) move the latest firmware into the card. reboot camera

feiticeir0, 2019-10-30:
@awannet what files should I put in the SD card ? I have a 1080p Home

ketpa, 2020-01-27:
After install yi-hack-v4 for 1080p camera, I can't add the camera with the yi app
How can I remove yi-hack-v4 ?

idanny76, 2021-05-30:
> 
> 
> After install yi-hack-v4 for 1080p camera, I can't add the camera with the yi app
> How can I remove yi-hack-v4 ?

Did you ever figure this out? Same problem.... Cant figure out how to remove yi-hack v4.