TheCrypt0/yi-hack-v4 #98: Hack don’t start

AngelofromTurin, 2019-07-05:
Hi have the camera in pic, i try to put in the file for yi 1080p Home , but dont work... isnt’s thise the right model? 
[image]

l4ci, 2019-07-07:
Same here. Tried yi 1080p Home - doesnt flash. Tried resetting multiple times. No change.

xeronex, 2019-07-20:
I've just tried to flash my 1080P home to day multiple times as well.. any updates?

phaxx, 2019-07-27:
+1

I've tried 4 different Yi 1080p Home cameras, couple of different memory cards. The firmware update just does not start.

DrArclight, 2019-08-14:
I've tried this with Yi Home 1080p model with firmware 4.5 and it's not starting the firmware update either.

ysinata, 2019-08-24:
just FYI i have the same issue of trying to flash yi home1080, but, the camera in the picture isnt it. U bought the wrong camera, I believe that is Xiaomi Mijia SXJ02ZM and i believe there is a different hack for that particular camera.

DrArclight, 2019-09-16:
My camera is definitely not the SXJ02ZM.  Mine uses the MCS313e chip.

ysinata, 2019-09-16:
I might be wrong about the exact serial number of your camera, but im 100% sure that urs is a "mijia" branded camera. From what i know, yi hacks only works on "yi" branded camera. "Mijia" and "fang" branded cameras each has different hack accordingly. And one more thing, u can find ur model number on that particular mijia camera on the foot stand. Reverse ur camera and look at the bottom. There should be some chinese character, and only few alphabets that u can read, thats the model number

DrArclight, 2019-09-18:
My camera is definitely Yi branded.  This is a brand new set of 5 cameras I just bought from Amazon a few weeks back.  yi Home 1080p branded.  Says it on the box, and on the cameras.  Even had to remove a little "yi" anti-tamper label from a screw hole to open it up.

ysinata, 2019-09-18:
Loll i just realized, my comment wasnt directed yo u. It was directed to @AngelofromTurin . In any case if yours is Yi camera, and you said is Yi Home 1080, the most likely scenario is u have the 6FUS version. Check the back of your camera, ur cam version is the first 4 digit. 

The bad news is, yi hack v3 and v4 doesnt work with one or two version of the yi home 1080 camera, 6FUS being one of it.

Sorry for ur problem, but, i am on the same boat as u 😅 still looking for different solutions atm