TheCrypt0/yi-hack-v4 #99: Find out camera version

l4ci, 2019-07-07:
Hey guys,

I am having trouble installing the yi-hack.

I wonder if it has to do with the quite new cameras (https://de.aliexpress.com/item/32818420214.html?spm=a2g0s.9042311.0.0.4df24c4d17VFgF).

Is this supported?

How do you even read out the correct version of the camera? I dont see it anywhere any you guys seem to read more ... like "24US, 5FCN ..." (https://github.com/TheCrypt0/yi-hack-v4/issues/74)

This is what I can read from the `diagnosis.txt` from the SD Card:

```
>>> Device Diagnosis <<<
+ cat /etc/os-release
NAME=Buildroot
VERSION=2017.08-gb77618e
ID=buildroot
VERSION_ID=2017.08
PRETTY_NAME="Buildroot 2017.08"
MODEL_COMMIT=7f43c64
COMMON_COMMIT=b77618e
XIAOMI_VERSION=3.5.1_0066
XIAOMI_BUILDNO=85  
#Build in Mon Apr 15 04:58:33 UTC 2019/ 64bf91150a4e by jenkins
```

edofullin, 2019-07-10:
Scan the qr code at the back of the camera, the first 4 chars represent the model. 
Yours is 04CF which, as you can see [here](https://github.com/TheCrypt0/yi-hack-v4/wiki/Supported-Camera-Models), is not supported

l4ci, 2019-07-10:
Alright thanks!

closed 2019-07-10 by l4ci