TheCrypt0/yi-hack-v4 #101: Restart/reboot camera via mqtt

niccolobusato, 2019-07-17:
single mqtt topic to trigger "reboot camera" button.

easy to use in Home Assistant or mqtt dash

jokerigno, 2019-08-23:
+1 on this. Sometimes the rtsp stream become unavailbe (while webserver works) so I need to enter and reboot. It would be great if this can be done via mqtt for example

bednim, 2020-02-17:
yes please on this too, or even a simple http link with direct reboot execution would do lots!

stevegoossens, 2020-03-06:
The `Reboot Camera` button calls "http://your_camera_ip/cgi-bin/reboot.sh"

I just loaded the URL in a browser to fetch it, and it returns an empty JSON object `{}` and reboots the camera.

In summary, you should be able to set something up, using the following info:

HTTP GET "http://your_camera_ip/cgi-bin/reboot.sh"
expect 200 response
(optional) expect application/json response with body `{}`

niccolobusato, 2020-03-19:
thank you for your work

 How can i add this on Home Assistant as a button?

stevegoossens, 2020-03-19:
you should be able to configure a Button card with the action to call a shell command:

```
# https://www.home-assistant.io/lovelace/button/
type: button
entity: camera.living_room
# https://www.home-assistant.io/lovelace/actions/#tap-action
tap_action:
  action: call-service
  service: shell_command.reboot_camera_living_room
  confirmation: true

shell_command:
  reboot_camera_living_room: "curl http://your_camera_ip/cgi-bin/reboot.sh"
```