TheCrypt0/yi-hack-v4 #102: Yi Outdoor and RTSP in HomeAssistant Connection refused

majkers, 2019-07-18:
Hi.
I managed to run RTSP i HA 0.96.0 but it suddenly stopped to work on Yi Outdoor. Even though it still works for my Yi Dome 1080p...

In HA log all I can see is:
`Connection to tcp://192.168.1.6:554?timeout=5000000 failed: Connection refused`

My configuration in `configuration.yaml` looks like this:
`stream:
ffmpeg:`
`camera:`
`  - platform: ffmpeg`
`    name: kamera_frontowa`
`    input: -rtsp_transport tcp -i rtsp://192.168.1.6/ch0_0.h264`

and for the card:
`camera_view: auto **BUT I ALSO TRIED live**`
`entity: camera.kamera_frontowa`
`camera_image: camera.kamera_frontowa`
`hold_action:`
`  action: call-service`
`  service: camera.snapshot`
`  service_data:`
 `   entity_id: camera.kamera_frontowa`
`    filename: >-`
`      /config/www/kamera/kamera_frontowa-{{ now().strftime("%Y-%m-%d-%H%M%S")`
`      }}.jpg`
`name: Front`
`show_state: false`
`type: picture-entity`

I can still access to it via yi app cause I have cloud enabled. It was working for a while but it stopped during the night.

owenworley, 2019-07-30:
Looks to be the same as what I am seeing - (With Yi Outdoor 1080p camera) Yi App works and so does rtsp, but asfter some amount of time (unpredictable amount) rtsp stops working (vlc, any other software trying to load the rtsp stream times out) - suggest linking this to isue #89

Scope666, 2019-07-30:
I have the same issue on Yi Outdoor 1080p and Yi Home cameras.  For now I've been leaving it off as eventually the camera becomes unresponsive and needs a reboot.

OLTSYS, 2019-09-09:
I have the same issue with the Yi_Dome Camera (720p). The camera need to be rebooted once a day, or homebridge cannot connect anymore using rtsp. :( ..

majkers, 2019-11-11:
Any news on that? It is really annoying .... it works for some time ane then stops with errors in HA like this:
Error demuxing stream: No dts in packet
error while decoding MB 54 21, bytestream -32
etc.

Bruteforce150, 2019-11-18:
Same issue on dome 1080.